Ultimaker/Cura #15062: Printed model with Cura 5.3 has lines in z-direction

InScene, 2023-03-27:
### Application Version

5.3

### Platform

Windows 10

### Printer

Ender 3 S1 Pro with Sonic Pad

### Reproduction steps

I have sliced and printed the following model in Cura 5.3 [Thingiverse Egg](https://www.thingiverse.com/thing:4812640?fbclid=IwAR3HP7bMX7UPPCIjnLl7YwNSOzucalk7VCRIJFZ-PiQb_PUy5_TNl7uZCJM)
Everything looked good in the sliced preview.

### Actual results

It was printed with two lines in each of the four directions.
I printed a cylinder with the same dimensions and the same settings in Cura 5.3 and it was fine and had no lines.
The problem only occurs with this Thingiverse model in Cura 5.3.

### Expected results

A print without lines in each of the four directions. (Of course, there is a line for the Z seam).
I switched back to version 5.2.2 and printed the same model with the same printer profile and settings. And this print was good and has no lines.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[image]
[image]
[image]
[image]
[image]
[image]

[Cura_project.zip](https://github.com/Ultimaker/Cura/files/11083045/Cura_project.zip)

Fritzo81, 2023-03-28:
Have the same issue!

discip, 2023-03-28:
Duplicate of #14811!

discip, 2023-03-28:
Seems to be a duplicate of #14811!

InScene, 2023-03-29:
@discip I do not believe that this is a duplicate of #14811. It may be the same cause, but the error behaves slightly differently. I did not see any errors in the slicer as is the case in #14811. The errors were only present in the print.

discip, 2023-03-29:
@InScene
These artifacts are barely noticeable in Cura.
In order to be able to perceive the "deformations" in Cura, you have to zoom in and look very closely.

Please refer to this comment [jagged paths](https://github.com/Ultimaker/Cura/issues/14811#issuecomment-1462937823).

Were you able to spot the _**'cuts'**_ in the marked areas?

InScene, 2023-03-29:
@discip Okay, then I guess you are right. Then I'll close this topic and refer to the other topic. 

**Closed due to duplicate of #14811**

closed 2023-03-29 by InScene