Ultimaker/Cura #15064: Failt to splice

BrettM56, 2023-03-28:
### Application Version

5.3.0

### Platform

windows 10

### Printer

Creality 3 Neo

### Reproduction steps

Click on open (envelope top left) dragged a STL file onto build plate in cura, It also happens if just click on the file which then opens Cura

### Actual results

It starts to splice then half way through came up with a error "Failed to splice"

### Expected results

To be able to drag the file or click on the file and loads and splices normally, it fails any files 2 out 5 files I try I only use STL Files

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[Flexi Factory Cat Planter_Drain holes.zip](https://github.com/Ultimaker/Cura/files/11085383/Flexi.Factory.Cat.Planter_Drain.holes.zip)

iccfish, 2023-03-28:
5.3.0 can not splice STL file, and curaengine process crashed due to error 'Illegal Instruction'.

[image]

dump file attached.

[CuraEngine.exe.8148.zip](https://github.com/Ultimaker/Cura/files/11088052/CuraEngine.exe.8148.zip)

MariMakes, 2023-04-05:
Hey @BrettM56,

Welcome to the UltiMaker Cura Github 🚀 
Sorry, it took us a while to get back to you 😞 

I'll need a little bit more information from you.
Do you have a project file for us? It contains the printer and settings we need for troubleshooting. 
To save a project file go to File -> Save project. Please make sure to .zip your project file. For big files, you may need to use WeTransfer or similar file-sharing sites.

You'll need to open Github in your browser to attach any files to your ticket
You can click the **view it on GitHub** link at the bottom of your email to open GitHub in the browser.