Ultimaker/Cura #15065: Cura 5.3.0 brakes artillery profiles

00raq00, 2023-03-28:
### Application Version

5.3.0

### Platform

Win10

### Printer

Artillery Genius/Hornet

### Reproduction steps

- slice [https://www.printables.com/model/6806-small-clip-for-herb-bags-other-packaging](clips stl) in cura 5.1 and 5.3 using defualt profiles for artillery printer (genius) 

### Actual results

- 5.3 generates unusable model, 5.1 is fine (screen) 
[image]

### Expected results

with default profiles it should be working in the same way ( or better, not worst) 

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Cura 5.3.0 brakes artillery profiles

MariMakes, 2023-04-05:
Hey @00raq00,

Welcome to the UltiMaker Cura Github 🚀 
Sorry, it took us a while to get back to you 😞 

Could it be that Print Thin Walls is disabled?
Can you try again with that setting enabled? 

If that's not it, I'll need a little bit more information from you.
Do you have a project file for us? It contains the printer and settings we need for troubleshooting. 
To save a project file go to File -> Save project. Please make sure to .zip your project file. For big files, you may need to use WeTransfer or similar file-sharing sites.

You'll need to open Github in your browser to attach any files to your ticket
You can click the **view it on GitHub** link at the bottom of your email to open GitHub in the browser.

00raq00, 2023-04-05:
> Hey @00raq00,
>
> Welcome to the UltiMaker Cura Github 🚀 Sorry, it took us a while to get back to you 😞
>
> Could it be that Print Thin Walls is disabled? Can you try again with that setting enabled?
>
> If [...]

Yeap with print thin walls enabled it works similar to 5.2.2 in this case. Unfortunately this feature is disabled by default so it is breading change. 

Prints with 5.3.0 with thin walls enabled is worst than 5.2.2. I can see gaps in prints.

MariMakes, 2023-04-05:
Hey @00raq00,

Thanks for the update. 

I'll need a little more information from you before I can bring it up with the team. 
Do you have pictures of the Gaps in the Prints? 
Can we also have a Project file? It has your printer and project settings. 
To save a project file go to File -> Save project.

00raq00, 2023-04-05:
> Hey @00raq00,
> 
> Thanks for the update.
> 
> I'll need a little more information from you before I can bring it up with the team. Do you have pictures of the Gaps in the Prints? Can we also have a Project file? It has your printer and project settings. To save a project file go to File -> Save project.

Currently i had to switch to Prusaslicer due to problems with 5.3.0 :-( it is really buggy version :⁠-⁠(

I will try find prints with gaps... Regarding to settings, 
Filament: 
generic pet (215 degrees) or generic petg (235 degrees)

Printer:
Artillery Hornet - standard quality (with 5.3.0 with thin walls enabled only)

00raq00, 2023-04-20:
> Hey @00raq00,
> 
> Can you help me help you? We cannot fix things if we don't know they are broken. I'll need some examples to show to the team so they can define a priority.
> 
> For that, I need some pictures that the quality decreased and a project file to analyze. If you can show that the same printjob is worse in 5.2 than it is in 5.3.
> 
> A project file contains the exact printer and profiles we need for troubleshooting. To save a project file go to File -> Save project.
> 
> You'll need to open Github in your browser to attach any files to your ticket You can click the **view it on GitHub** link at the bottom of your email to open GitHub in the browser.
> 
> I hope you understand that at UltiMaker we don't have the Artillery machines to test on and for introducing and updating these profiles we rely on our community members to help.

I cannot do it now. It is not reason to close another issue of worst 5.3.0 version of CURA
 
You have all information to reproduce it without printer and my help. Get a grip.

nallath, 2023-04-20:
Woah. Please change the tone a bit, we are trying to help you. I'd like to point you to the code of conduct, which you can find here: https://community.ultimaker.com/guidelines/. I understand that you're frustrated, but we need your help to get to the bottom of this. Comments like this or the ones made on https://github.com/Ultimaker/Cura/issues/15269 are not constructive or helpful. 

The ticket, as the bot helpfully told you, was closed because we asked a question that wasn't answered for 14 days. 

We asked you for some extra info, and we need that to get a grip on this issue. It's okay if you don't want to provide this, but it also means that we won't continue without it.

00raq00, 2023-04-20:
> Woah. Please change the tone a bit, we are trying to help you. I'd like to point you to the code of conduct, which you can find here: https://community.ultimaker.com/guidelines/. I understand that [...]

Yeah I can understand your point of view. The problem is that in this issue i provided all information to reproduce issue, printer name, printer configuration, url to model and screenshots from 5.1/5.2.2 and 5.3 that show issue. It is not required to print model to reproduce issue. 

5.3 version causes many new huge issues (before it had many but small issues) 
e.g. https://github.com/Ultimaker/Cura/issues/14958 shows gaps issue. Another one (i cannot find right now) shows that layers are not aligned correctly.

MariMakes, 2023-05-05:
Hey @00raq00,

I took another look at your report. 
Where you mention that Cura _generates an unusable model,_ I believe you mean that parts of the model are connected where they should not be. 

[image]

The setting _Print Thin Walls_ also counts for thin gaps. Enabling this setting will let you print this model correctly. 
Most printers have the _Print Thin Walls_ setting enabled by default, that might be why you are the first person to report this. 

I'll ask the team why the behavior with the setting turned off changed between Cura 5.1 and Cura 5.3. I suspect it might have something to do with the introduced fixes for the crashes. 

I expect that if it's not a quick fix that it will not get a high priority to implement because there is a good workaround in place. 
I hope this helps 😇

MariMakes, 2023-05-08:
Hey @00raq00,

Quick update from our side.
I discussed your issue with the team and we believe we will not work on this anytime soon and will therefore defer this issue.
Since Cura is open source, you or anyone else is more than welcome to work on this issue and create a pull request yourself.

I'll be closing this issue. I hope you understand.

closed 2023-05-08 by MariMakes