Ultimaker/Cura #15069: Lack of Basic Travel Optimizations

FPrefect20, 2023-03-28:
### Application Version

5.3.0

### Platform

Windows 10

### Printer

Any Printer

### Reproduction steps

1.  slicing simple 2-wall object
2.  watching the printing process

### Actual results

The actual printing order for layers is "simply every layer the same way" (without calculating on the next level)
wall#1; travel; wall#2; travel;  z+1;  [...]
no optimization can be found.
Setting "optimize wall printing order" has no effect.

### Expected results

Simple optimization printing 2 layers at a time for each wall before retraction/travelling
This could save nearly 50% of the RETRACTION + TRAVELLING.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[wall_test_STL.zip](https://github.com/Ultimaker/Cura/files/11094001/wall_test_STL.zip)
[image]

MariMakes, 2023-04-05:
Hey @FPrefect20,

Thanks for your report 👍
I'm not sure if I understand your bug correctly. 

Right now Cura starts the layer always at the same point (defined by Layer Start X and Layer Start Y)
So in this Gif the layer always start on the right wall. 
[image]

You would prefer the layers to start on wall Right, than on wall Left, than on wall Right, than on Wall Left repeat?

I believe that has always been the case, would you like me to change your bug to a feature request instead?
For feature requests we often ask the following questions: 

- Who do you think will benefit from this? Is everyone going to benefit from these changes? Or specific kinds of users?
- Can you give a clear and concise description of any alternative solutions or features you've considered. If possible, think about why these alternatives are not working out.

FPrefect20, 2023-04-06:
Hi @MariMakes , thanks for your answer! 
If CURA really didn't hide this print optimization in the settings, then: "yes", please change my "bug" to a corresponding feature request. Thank You!

IMHO everybody will benefit from this (better quality @ smaller printing time)
But the 'optimization' should work automatically. 
i don't think that any kind of 'manual setting' (manually stating the printing order of each layer) makes sense; especially printing more than 2 separated elements (walls)

What you mean by "alternatives"? stop using "CURA" and trying alternative slicing tools?
In CURA 5.3 i tried every combination of the travel settings plus "optimize wall printing order", "z-seam alignment".

closed 2023-11-03 by github-actions[bot]