Ultimaker/Cura #15072: Layer Count error

GregValiant, 2023-03-29:
### Application Version

5.3.0

### Platform

Windows 11

### Printer

Creality Ender 3 Pro

### Reproduction steps

Load a model that tapers to feather edges on the top
Slice.
The preview shows 711 layers.  The gcode has layers 0 to 710.

### Actual results

The Gcode file shows a ";LAYER_COUNT:" of 718 layers.
The top layer in the gcode is 710 so it does match the 711 in the preview.

### Expected results

I've been playing with post processors and needed to grab the "Layer Count".
The model in the attached project file is 143.6mm tall and at .2 layer height that would indeed be 718 layers.  Because of the feather edges, the top of the model doesn't slice and so it is only 711 layers tall (as the preview shows).
This happened in 4.13.1 as well.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[LayerCountBug.zip](https://github.com/Ultimaker/Cura/files/11096126/LayerCountBug.zip)

MariMakes, 2023-04-05:
Hey @GregValiant,

Always good to see you 👋 😊 

[Here I learned that the layercount in Cura starts at 0 and counts up. ](https://community.ultimaker.com/topic/42890-ultimaker-cura-53-beta-released/page/2/#comment-320059)
We even updated the copy for pause at height to correctly reflect it https://github.com/Ultimaker/Cura/commit/84bb820e336bd293e7552b381df2bb9ea94391be

I took a look at your model and I'm seeing the correct behavior. The last layer is Total in the preview -1.
[image]

Are you sure you shared the correct project file because you mention 711 layers and I'm seeing more.

GregValiant, 2023-04-05:
Hi @MariMakes.
Yes, there was a change in Pause at Height.  Remco ran with it but I think I wrote it.

This model is for demonstration.  It is 150mm tall.
When sliced at 0.10 layer height the math says 150 / 0.10 = 1500 layers.  At 0.4 Line Width the Cura preview shows 1475 layers.  That's about right given that the top of the model is too narrow to print at 0.10 and 0.4.
The top layer in the gcode is 1474 which converted to base1 is indeed 1475 layers.
This snippet is from the same gcode file:
G92 E0
G1 F2100 E-6
;LAYER_COUNT:1500
;LAYER:0

So it appears that the "LAYER_COUNT" is the theoretical count, rather than the number of layers in the file.  That is exactly what happened with that original project file.  The LAYER_COUNT at the end of "data[1]" doesn't match the number of layers actually in the file.
[SharpCone LAYER_COUNT.zip](https://github.com/Ultimaker/Cura/files/11160511/SharpCone.LAYER_COUNT.zip)

Back to the original project file....
This is from the end of the gcode file generated from the project file I submitted with this report:
;TIME_ELAPSED:24490.075311
;LAYER:713
that is indeed the top layer so with the settings used there are 714 layers in the file.
But this is from the same file at the beginning:
;LAYER_COUNT:718
;LAYER:0

It isn't right.

If I reduce the base diameter of the "sharp cone" to 10mm then the gcode file has 1449 layers but the LAYER_COUNT still insists there are 1500.

This error occurs at least back through 4.13.1.

MariMakes, 2023-04-06:
Hey @GregValiant,

Thanks for the clarification!
I clearly have misread your ticket, sorry about that 😅 

I wasn't aware of the LAYER_COUNT. 
I took a look and indeed the LAYERCOUNT does not match the one in preview. 
[image]

I'll bring it up to the team to see what they can do to improve it.  💪

closed 2023-04-14 by rburema