Ultimaker/Cura #15075: slicer engine freezing/stopping

zero12003, 2023-03-29:
### Application Version

5.3.0

### Platform

windows 10

### Printer

ender 3

### Reproduction steps

attempting to slice a cindewings rose dragon 3d model and slicer sits idle for extended period before popping up an error saying the slicing engine has stopped and suggesting it be reported as a bug. Previous version of cura works fine just has a long processing time with the same model.

### Actual results

force close cura

### Expected results

generate an stl file.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

cinderwings 3d file only has this isue with current version of cura.

MariMakes, 2023-04-05:
Hey @zero12003,

Welcome to the UltiMaker Cura Github 🚀 
Sorry, it took us a while to get back to you 😞 

Have you tried reverting to the default settings?
I can slice the model with default settings in Cura 5.3
[image]

If that's not it, I'll need a little bit more information from you.
Do you have a project file for us? It contains the printer and settings we need for troubleshooting. 
To save a project file go to File -> Save project. Please make sure to .zip your project file. For big files, you may need to use WeTransfer or similar file-sharing sites.

You'll need to open Github in your browser to attach any files to your ticket
You can click the **view it on GitHub** link at the bottom of your email to open GitHub in the browser.

zero12003, 2023-04-07:
thanks for getting back with me.  Here is the project file with default settings. I am still having issues with it not working/crashing cura whenever I try to slice this model and it is only this model having problems it works fine with my other slicing program. Thank you for the help.
[rose dragon project.zip](https://github.com/Ultimaker/Cura/files/11182093/rose.dragon.project.zip)

MariMakes, 2023-05-03:
Hey @zero12003,

Thanks for sharing the project. 
I took another look and I noticed that you have a custom printer/profiles. 
I don't have those so I'm getting this error:
[image]

Could it be related to these custom settings? 
Have you tried reverting to the default settings?
Have you tried moving the model away from the center of the buildplate and rotating by 90 degrees?

zero12003, 2023-05-03:
Hello,

Thanks for getting back to me. The custom profile is just a copy of the
ender 3 profile with a different name so I can tell which printer is being
used. I did try reverting everything back to standard settings to no avail.
I have found that reverting to cura version 5.2 works flawlessly with this
project but as soon as I enable the 5.3 upgrade the slicer simply freezes
and will do nothing.

Thanks for your help.

John