Ultimaker/Cura #15076: Bad slicing result

GMagician, 2023-03-29:
### Application Version

5.3.0

### Platform

Windows 11

### Printer

Custom Bear 2.1

### Reproduction steps

Just slice attached project

### Actual results

There are some weird, uncompleted, top layers

### Expected results

Correctly filled top layer

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[image]
[Lid.zip](https://github.com/Ultimaker/Cura/files/11101171/Lid.zip)

GMagician, 2023-03-29:
Not sure log may help here

MariMakes, 2023-04-05:
Hey @GMagician,

Thanks for your report 👍

This seems to be a duplicate of https://github.com/Ultimaker/Cura/issues/15028
There we learned that you can resolve it by increasing the Skin Expand Distance, that also works for your case. 
[image]

We'll still discuss it with the team because it might be an indicator that something else is broken.
I'll be closing this issue as a duplicate but you can follow the progress here: https://github.com/Ultimaker/Cura/issues/15028

closed 2023-04-05 by MariMakes