Ultimaker/Cura #15078: Cura 5.3 skips every other layer on second extruder Prime Tower

Erin-Stephan, 2023-03-29:
### Application Version

5.3

### Platform

MacOs 12.0.1

### Printer

Essentium Inc. (Custom printer)

### Reproduction steps

Slice a dual head print with prime tower enabled.
Preview slice.
Every other layer of internal second extruder is missing. (and not printed when I try to print)

### Actual results

Every other layer of internal second extruder is missing. (and not printed when I try to print)
[image]
[image]

### Expected results

The second extruder should be printing material on every layer of the prime tower.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[EHSE2_PrimeTowerMissingLayersTest1.3mf.zip](https://github.com/Ultimaker/Cura/files/11103617/EHSE2_PrimeTowerMissingLayersTest1.3mf.zip)

Erin-Stephan, 2023-03-29:
linking issue #14606 that I think is the same issue but it got closed because it timed out

Erin-Stephan, 2023-03-29:
Also this may be related to the issues in this ticket -  #13475 - where the prime tower order was going to be changed, either as a result of the main fix or the secondary quick fix by @rijkvanmanen. 

T0 is still in the old order (prime, part, prime, part) T1 is almost to the new order (prime, part, part, **Prime**) but that **prime** is not printing on the tower.

TheThomasD, 2023-03-30:
Just my two cents: I think the general behavior of having one extruder building the outer wall of the prime tower on every layer and the other extruder only using the prime tower when it is needed (every second layer) is quite a good approach. I select the "cheaper" filament as the first one so I save more of the expensive one in the second extruder. Also, if you have a mixing extruder, using the darker color filament in the first extruder (less priming volume needed), you can even save filament there. The oder that I imagine (and that seems correct to me) is:

1. print first layer (with prime tower brim and e.g. a skirt) with E0
2. print first layer with E1 (e.g. with skirt as well as a priming method)
3. print second layer with E1 (no need for switching)
4. prime with E0 on second layer, print second layer with E0
5. print third layer with E0, prime with E0 on third layer (needed for prime tower stability)
6. prime with E1 on third layer, print third layer with E1
7. print fourth layer with E1
8. prime fourth layer with E0, print fourth layer with E0
9. ... and so on.

As you can see, E0 always builds the outer prime tower wall (for stability), but E1 only primes every second layer. I can understand that you might _want_ to prime on every layer, but you don't have to, right?

Erin-Stephan, 2023-03-30:
@TheThomasD I agree with your ordering except I see the need for priming E1 at "step 7" for two reasons. 

1. Printing every other E1 prime is causing the E1 strands to not adhere and get drug into the part. See pic below
2. I would like E1 to prime after its second layer (step 7 above) to be able to cool down during the prime tower to prevent oozing. Having this happen at step 7 allows E1 to cool mostly during the prime tower instead of the part (using final temp and cool down speed) so I like that method. If it cools during the part, you could be sacrificing some strength and could introduce a defect. 

[image]

MariMakes, 2023-03-31:
Hey @Erin-Stephan,

Thanks for your report 👍
This was also reported internally 😉 

It should be resolved in the recently merged https://github.com/Ultimaker/CuraEngine/pull/1798/files
It was one of the acceptance criteria for it to be merged. So I expect this to resolved in the next version of Cura.

I just doublechecked in a nightly version we released earlier today and it looks like this. Priming every layer. 
[image]

If you want to take a look, you can do that here: https://github.com/Ultimaker/Cura/discussions/15098

Can you let me know if this was what you were looking for so I can close the bug?

glx-seven, 2023-04-02:
Nice this issue seems to be finally solved :), it has been around since like the beginning of time (at some point I was told this is the intended behaviour).

> As you can see, E0 always builds the outer prime tower wall (for stability of the tower), but E1 only primes every second layer. I can understand that you might _want_ to prime the second extruder on every layer, but you don't have to, right?

Actually it doesn't do that in every case (see the screenshot below), also it leads to very unreliable results since the inner priming layers don't really adhere anywhere and can end up being dragged into the print causing it to fail.

[image]

GoguyT3D, 2023-04-03:
@MariMakes Is there an installation file I'm missing or am I going to need to overwrite current config files, I'm not sure how to implement the code the link above led me to? (https://github.com/Ultimaker/CuraEngine/pull/1798/files)

Erin-Stephan, 2023-04-03:
@GoguyT3D My understanding is that we have to wait until they release the next version of cura to get this feature but maybe theres a way to implement it earlier? I'm not sure

GoguyT3D, 2023-04-03:
> @GoguyT3D My understanding is that we have to wait until they release the next version of cura to get this feature but maybe theres a way to implement it earlier? I'm not sure

#15098 mentioned earlier seems to be the alpha "nightly" version not tested yet, I'm under the impression it's in there

wawanbreton, 2023-04-04:
Hi everyone,

I'm glad to see that the prime tower has finally been improved ! However, it is quite a regression for people who use a printer with a dual Y extruder (two filament inputs, a single nozzle) because we actually use the prime tower as a "purge tower" and we have to make it very large so that the purge is complete (e.g. when you switch from black to white). With the current implementation, it uses too much filament (and time) as what is really required, and it will be even more the case with this fix.

I have implemented a proposal for an optimization of the prime tower, but have not made a pull request yet because I wanted to have some feedback before I did. This version fixes the problem of having layers printed with nothing under, and it also optimizes the amount of filament/time used. @MariMakes would you please take a look at it ? Discussion topic is here : #6909 and implementation is on my fork : https://github.com/wawanbreton/Cura/tree/optimized-prime-tower and https://github.com/wawanbreton/CuraEngine/tree/optimized-prime-tower

wawanbreton, 2023-04-04:
@GoguyT3D Good to see that you could finally print your part !

I'm not sure the prime tower mode selection can be printer-dependant. It also depends on the material used (whether they stick to each other or not). What I think is that the "optimized-consistent" version could maybe replace the actual one, because it actually does the very same job, but extrudes a bit less filament when not required. The "optimized" one could be activated manually as an option, if you have a Y extruder and known for sure that all used filaments stick to each other.

MariMakes, 2023-04-05:
Hey @wawanbreton,

Thanks for tagging me! 
I wasn't aware of this branch and it looks super cool. 
Amazing work 💪 

I asked our Print Profile Experts to do some in-house experiments.
But I don't believe we have mixed nozzles. 

@nallath can you help out when he gets stuck with a Pull Request?

wawanbreton, 2023-04-05:
> I wasn't aware of this branch and it looks super cool. Amazing work :muscle:

Thanks :blush: 

> I asked our Print Profile Experts to do some in-house experiments. But I don't believe we have mixed nozzles.

That's great !! From what I can tell, neither Ultimaker nor Makerbot printers have mixed nozzles, but it would still be very interesting to see how this tower performs with two extruders and a small prime tower. You must also have more materials than me, and you can try to mix them.
On my side, I will keep doing tests with my Y extruder. Tell me if there are some specific configurations you would like me to test. 

> @nallath can you help out when he gets stuck with a Pull Request?

Just to know, what is your convention to make a pull request ? Should I do it right now and keep improving the code, or wait for it to be fully stable before ?

MariMakes, 2023-04-06:
Hey @wawanbreton,

I found more excited Cura developers willing to help 😊 

They told me, if you start a draft Pull Request they can start helping with improving your code where it's needed. It's also the best way to collect your insights and tests. Which are now spread across a few tickets I believe 😉 

[This is an example of a pull request that ended ](https://github.com/Ultimaker/CuraEngine/pull/1282) up becoming the Xmas Tree Support Alpha, but [pull requests like these also ](https://github.com/Ultimaker/CuraEngine/pull/1757) make it into releases. 

The PR also comes with a template that you can complete that can help you with structuring your information. Good luck!

wawanbreton, 2023-04-06:
@MariMakes Thank you so much... I didn't know a pull request could be tagged as a draft. I'll take some time to properly make it !