Ultimaker/Cura #15081: proyect

Titohusky, 2023-03-29:
### Application Version

5.3

### Platform

windows 10

### Printer

Ultimaker 

### Reproduction steps

el soporte de arbol falla

### Actual results

mal resultado

### Expected results

ninguna

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

no funcionan los soporte de arbol de ninguna manera

Titohusky, 2023-03-29:
espero que mejore

MariMakes, 2023-04-05:
Hey @Titohusky,

Lo siento, no puedo hablar español, pero creo que puedo ayudarte.

Welcome to the UltiMaker Cura Github 🚀 
Sorry, it took us a while to get back to you 😞 

The new Cura 5.3 Stable contains the fix for this for a number of slicing crashes. 
Could you test if that's the case for your model as well? 
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.3.0

Have you tried reverting to the default settings?
Have you tried moving the model away from the center of the buildplate and rotating by 90 degrees?
Can you check if your mesh is faulty with the Meshtools Plugin? You can get it here: https://marketplace.ultimaker.com/app/cura/plugins/fieldofview/MeshTools

If that's not it, I'll need a little bit more information from you.
Do you have a project file for us? It contains the printer and settings we need for troubleshooting. 
To save a project file go to File -> Save project. Please make sure to .zip your project file. For big files, you may need to use WeTransfer or similar file-sharing sites.

You'll need to open Github in your browser to attach any files to your ticket
You can click the **view it on GitHub** link at the bottom of your email to open GitHub in the browser.

closed 2023-04-20 by github-actions[bot]