Ultimaker/Cura #15082: New Top/Bottom Pattern

brokesc, 2023-03-29:
### Is your feature request related to a problem?

I've noticed that with some picky filaments the first top or bottom layer can be...spotty...including places where filament actually drops into the support hollows on the first, and even the second, pass.

### Describe the solution you'd like

For the top layers, having the line pattern criss cross by layer (so one layer goes NE to SW and next goes NW to SE assuming a NS/EW square) would effectively mean each layer is supporting the one above it while when the lines in each layer go parallel with each other, a sufficient gap just becomes a place where that pass is unsupported potentially increasing the size of the gap!

For that matter, if you really want to go crazy, allow users to specify the angle of rotation between each ceiling/floor layer (30 degree/60 degree/90 degree occur to me)

### Describe alternatives you've considered

Another possibility that occured to me is the option to have an offset in the fill pattern between layers (same pattern just say...a 0.2 mm shift on the x, y, or both axis). This would ultimately result in a honey comb type pattern where the voids aren't columns but rather are cells increasing the strength for exactly the same filament usage and reducing the amount of ceiling which isn't closely supported by a lower layer (causing a dip instead of a sinkhole...)

for that matter, the angle of rotation option proposed for the celling/floor above could also be applied to infill...

### Affected users and/or printers

These options would potentially be of benefit for all users.

TheThomasD, 2023-03-29:
Not sure whether this is what you want, but you can specify `Top/Bottom Line Directions`.

[image]

By doing so, the first layer is printed 30 degrees, the next 60 degrees, ... (you can add a lot of numbers there...) Cura starts the top and the bottom layer both with the first number and then goes to the next, so it's not possible to have a different list for top and bottom.

That said, might I suggest that you check whether you can print the filament at a lower temperature? It sounds to me that it is becoming too liquid and reducing the temperature could maybe improve that.

Cheers
Thomas

brokesc, 2023-03-29:
But can you specify for each layer?

TheThomasD, 2023-03-29:
Well, you define a list of angles there. The first angle is used for the first layer of the top/bottom, the second for the second layer, and so on. So yes, for each layer you can specify a direction. Please give it a try to see what happens. Defaults are often [45,135], but you can e.g. try [45,90] to see the difference. If Cura reaches the end of the list, it will start at the beginning again. So in the [45,90] example the third layer will be 45 degrees again.

brokesc, 2023-03-30:
Sweet. Thx

TheThomasD, 2023-03-30:
Maybe you want to close this issue then?

brokesc, 2023-03-30:
Yes, I would. Thx again.

closed 2023-03-30 by brokesc