Ultimaker/Cura #15083: line start speed.

TjeerdvanDijk, 2023-03-29:
### Is your feature request related to a problem?

When beginning new a line on the build plate the nozzle moves at a constant speed. When build adhesion is not absolutely perfect from the starting point of the line, it won't adhire. The slightest tension it develops during the first 5mm will make it come los. It will either curl up or get dragged away by the nozzle.

This can especially be an issue when using a adhesive (like a glue stick) that hardens after some time.

The most obvious solution is to slow down the first layer/first layer wall speed. But for bigger models that can be quite time consuming. 

### Describe the solution you'd like

Being able to control the speed of X mm after a travel move at the first layer.

### Describe alternatives you've considered

slowing down the first layer wall speed, it works but the entire wall has to slow down.

### Affected users and/or printers

It is easily testable by applying gluestick to the build plate of any printer and leaving the bed off. For my test i used the default build plate of a ender 3

TheThomasD, 2023-03-29:
Just an idea (not sure if this could help you): maybe you can set the acceleration to a very low value for the initial layer. That way, large models would still print quite fast (if the top speed can be reached), but the start and stop of each line would be very smooth.

Another option: you could try 3Dlac. It is similar to hairspray, but it's way more sticky (IMHO). I use it quite often.

pietryna, 2023-03-30:
I can also add, that I "suffer" the same from the same problem. 

My solution is, first, to print really "thick&heavy" prime line (default for Creality but is quite handy) at the side of the bed and then just print 3+ skrit lines to stabilize the flow (first 10-15cm of is visibly thiner)
Then I do not see problem you're describing when head moves to print actual model while prime line and usually start of skrit is affected.

TheThomasD, 2023-03-30:
I do the same as @pietryna. That has worked on all my printers so far and I didn’t even know that this has an effect on the line sticking on the plate or not.

TjeerdvanDijk, 2023-03-30:
I don't think the problems you both have are the same, i print a skirt of 3 lines and by bed is 35x35cm. I also have to add to the problem that my print head has to travel quite a bit sometimes and during that travel move the pressure changes, overal there isn't a lot of pressure in my setup to start with.

See the picture here below, You can see some places it started a line but it didn't adhire, everything else is adhired and has a strong bond. 

[image]

TheThomasD, 2023-03-30:
Just to be sure you checked everything else: do you use bed leveling with a probe? That often helps with uneven beds. Is this maybe an issue? I only use glass beds (except for the belt printer) as I had some issues with the sticker-based beds being to uneven, e.g. due to small air bubbles under the sticker or an uneven heating plate.

As for the pressure: are you using linear advance as well as well-tuned e-steps and retraction settings? With that, you shouldn’t have any issues with pressure.

also, can you maybe explain why the pressure changes in your nozzle during a travel? Do you experience oozing? If so, maybe you should fight that first.

TjeerdvanDijk, 2023-03-30:
Good point. I use klipper and did an analysis on the bed flatness. The biggest difference there is 0.06mm, but I actually also get underextrusion at the highest position. I tried it with mesh leveling on and off, it didn't make a difference. 

For the bed i applied the sheet myself to the spring steel plate and I am sure there are no bubbles, it actually isn't  the default ender 3 build plate but it is exact the same material and texture. 

I did calibrate esteps and retraction, it is a direct drive setup and retraction is set to 1mm, I did experiment with the retraction length and speeds, retraction towers show excellent results. Changing those setting didn't make a difference for my problem. 

I did tune linear advance multiple times, but either turning it on or off doesn't show a difference at all.

Good to know is that I have absolutely no problem printing a 10x10cm cube, or anything that size. No over or under extrusion. 

Basically when there is a travel above 7cm I instantly get underextrusion, I doesn't matter at which layer. I tried setting extra prima amount to a high value and it does create a blob but then instantly after it underextrudes, strange isn't it. My theory is that the filament suddenly has more time to soften in the nozzle and therefore loses pressure. (travel speed is 150mm/s) 

I am using a high flow hotend (mosquito) and when I push and retract filament it seems very responsible, notable is that it doesn't require force to do that. Lower temps also didn't solve my issue. When I do a cold pull everything seems very normal. It even came down to disassembling the nozzle to

TjeerdvanDijk, 2023-03-30:
Good point. I use klipper and did an analysis on the bed flatness. The biggest difference there is 0.06mm, but I actually also get underextrusion at the highest position. I tried it with mesh leveling on and off, it didn't make a difference.

For the bed i applied the sheet myself to the spring steel plate and I am sure there are no bubbles, it actually isn't the default ender 3 build plate but it is exact the same material and texture.

I did calibrate esteps and retraction, it is a direct drive setup and retraction is set to 1mm, I did experiment with the retraction length and speeds, retraction towers show excellent results. Changing those setting didn't make a difference for my problem.

I did tune linear advance multiple times, but either turning it on or off doesn't show a difference at all.

Good to know is that I have absolutely no problem printing a 10x10cm cube, or anything that size. No over or under extrusion.

Basically when there is a travel above 7cm I instantly get underextrusion, I doesn't matter at which layer. I tried setting extra prima amount to a high value and it does create a blob but then instantly after it underextrudes, strange isn't it. My theory is that the filament suddenly has more time to soften in the nozzle and therefore loses pressure. (travel speed is 150mm/s)

I am using a high flow hotend (mosquito) and when I push and retract filament it seems very responsible, notable is that it doesn't require force to do that. Lower temps also didn't solve my issue. When I do a cold pull everything seems very normal. It even came down to disassembling the nozzle to inspect any issues, but everything looks normal.

Here is a picture that shows this underextrusion after a long travel, i wonder what I could do, yea I believe you are right that I should try to fight this issue, but slowing it down that first 5mm is a way to fight it

[image]

TjeerdvanDijk, 2023-03-30:
I forgot to mention that i had the exact same problem with holes, and i solved the issue by slowing it down using Small Hole Max Size/Small Feature Speed. So that is basically where i got the idea from