Ultimaker/Cura #15084: [5.3.0] Model is incorrectly sliced (some layers are larger than they should be)

TheThomasD, 2023-03-29:
### Application Version

5.3.0

### Platform

Windows 10

### Printer

Artillery Sidewinder X1 (but similar issues with other printers)

### Reproduction steps

1. load the attached model
2. note that the inner walls are completely flat
3. slice the model laying "flat on the back" (see screenshots. I didn't find any setting that have an influence on the issue, e.g layer hight, mesh fixes, ... except for changing the orientation, which is not what I want)
4. look at the results (in Cura and in the produced gcode)

[image]

### Actual results

Some layers are shifted towards the inside of the model (see the inner walls shining through). Actaually, if you look closely, it is not only one layer, but multiple layers (you can see that on the edges).

[image]
[image]

### Expected results

flat walls as in the model (see attached stl).

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[Mold.zip](https://github.com/Ultimaker/Cura/files/11106390/Mold.zip)

TheThomasD, 2023-03-30:
One thing that I noticed: the larger layers start where the horizontal holes begin and they end where the top of them is closed. It seems these things are connected.

TheThomasD, 2023-03-30:
I found a workaround:  I closed the holes from the outer side of the model with 0.1mm wide "walls". Then, when I sliced it again, the holes are there as expected (I guess Cura removes these very fine lines) and the model is correctly sliced.

[image]

Model: 
[Mold2.zip](https://github.com/Ultimaker/Cura/files/11113331/Mold2.zip)

MariMakes, 2023-04-05:
Hey @TheThomasD,

Thanks for your report 👍

Your issue reminds me of some issues we have related to Hole Horizontal Expansion. 
In your original model, not the updated one, the expansion stops when the layer sees a C instead of an O and doesn't apply the horizontal expansion. 

That should be fixed with the introduction of the Maximum Hole Horizontal Expansion which is implemented here: https://github.com/Ultimaker/CuraEngine/pull/1768. You can expect it in the next release. 

To me, it doesn't seem like a bug in Ultimaker Cura anymore, because there is a solution in place. 
I'll close the issue.

TheThomasD, 2023-04-05:
Hi @MariMakes,

thank you for looking into this. However, I don't really get what you men by "the layer sees a C instead of an O". Can you please explain this a little bit more?

Also, why do you think it is related to hole horizontal expansion? My understanding was that this expansion is applied to vertical holes, but they are expanded in horizontal direction. Now, my holes are vertical. Does the expansion also apply to these? If so, why would the walls of the rest of the model be expanded instead of beeing shrinked (i.e. making the hole larger)? Did you test the changes you mentioned against my model or was this just a guess?

I just want to make sure this isn't lost if it is not covered by the fix you mentioned.

Thanks
Thomas

MariMakes, 2023-04-06:
Hey @TheThomasD,

The tooltip for Horizontal Hole Expansion reads: _Amount of offset applied to all holes in each layer. Positive values increase the size of the holes, negative values reduce the size of the holes._
If you need any more detail, [this is how the setting is described in the Settings Guide. ](https://github.com/Ghostkeeper/SettingsGuide/blob/master/resources/articles/shell/hole_xy_offset.md)

Cura can only recognize a hole that lives on the same horizontal plane. 
So in a model that looks like this:
[image]

This would be considered an O
[image]

This would be considered a C
[image]

In the intersection where an O becomes a C, you can see that Horizontal Expansion stops being applied and you see a kind of dent. 
[image]

Does this explain the issue better?
If you want to experiment for yourself, here is the model I used [HoleHorizontalExpansion.zip](https://github.com/Ultimaker/Cura/files/11170022/HoleHorizontalExpansion.zip) for this.

TheThomasD, 2023-04-06:
Hi @MariMakes ,

thank you for the detailed explanation! I was confused with the holes in my model and the holes that appear on each layer. Now I get it: The holes in my model open up the "hole" (the cavity) in my model to the outside. As in the explanation you referenced:

> This setting compensates for that by making all holes slightly bigger. Unlike [Horizontal Expansion](https://github.com/Ghostkeeper/SettingsGuide/blob/master/resources/articles/shell/xy_offset.md) this only affects closed holes though. If there is even a tiny opening on one side (horizontally, on the same layer) then that part won't be considered a hole and won't be affected by this setting.

As the cavity hole is not a hole anymore (it is opened by the horizontal holes to the outside of the model), in that area the horizontal expansion is not applied, whereas in the other areas it actually is applied. So what I was seeing was not a wall that was enlarged, but actually this was the actual size. The layers below and above where just expaned.

This also explaines why closing the wholes with a thin layer corrected the slicing. Now, the cavitiy was closed on all layers, so all layers got expanded.

Thank you again
Thomas