Ultimaker/Cura #15087: Cura is ignoring Z axis settings.

nekoshade, 2023-03-30:
### Application Version

5.3

### Platform

windows 11

### Printer

ender 3

### Reproduction steps

while uncheking "drop down model" and rising it slightly from the bed, the following happens.

[image]

blue line is where the model should be printed, but preview clearly shows it being printed from the bed instead.
happens with either tree and normal support types.

### Actual results

cura is generating a preview (not tested in a actual printer) that is far from the user input.

### Expected results

it should rise the model and place supports below.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

zip file of the 3mf file.
https://drive.google.com/file/d/1jJMg0VZwdsiLiyZne2JQbQHJjM4X3mkj/view?usp=share_link

Asterchades, 2023-04-03:
I believe this is a change made with 5.3.0, but I'm not sure if it's a deliberate change. It certainly doesn't seem like it.

If you'll humour me, try disabling `Generate Supports` and then disable the `Remove Empty First Layers` setting under `Mesh Fixes`. Note that this setting is completely absent in the event supports are present (generated or user-specified), however it is still being applied despite its status. Restore the supports afterwards and try slicing again.

Apologies for the delay in response.

nekoshade, 2023-04-05:
It worked like it should with your guidance.
[image]
but those two options contradcting each other, and being very stashed away is something that still should be adressed.

Asterchades, 2023-04-05:
That's pretty much what I expected.

I'll flag this for investigation. It's an undocumented change from 5.2 and feels like it shouldn't be happening, but ultimately it's the main dev team who has the final say.