Ultimaker/Cura #15088: Skirt is printed AFTER support on first layer

Edocecrous, 2023-03-30:
### Application Version

5.3.0

### Platform

Linux 64

### Printer

Anycubic Predator

### Reproduction steps

Load model
Hit slice

### Actual results

The model is sliced, but the order is not correct, first the support is printed, then the skirt, then the model on the first layer.
[image]

### Expected results

In Cura 5.2.1 it works correctly,
Does the skirt first,
then the support (tree support)
then the model

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[NanoSD15fdmE.3mf.zip](https://github.com/Ultimaker/Cura/files/11109318/NanoSD15fdmE.3mf.zip)

MariMakes, 2023-04-05:
Hey @Edocecrous,

Welcome to the UltiMaker Cura Github 🚀 
Sorry, it took us a while to get back to you 😞 

Thanks for your report 👍
That definitely looks like a bug on our side. 
It might be related to whatever introduced https://github.com/Ultimaker/Cura/issues/14983 in Cura 5.3

I'll bring it up to the team to see what they can do to improve it.  
Fingers crossed 🤞

TommySchwartz, 2023-07-20:
I have the same problem in Cura 5.4.0, Windows 10.
Support prints before skirt.

MariMakes, 2023-07-27:
Hey @Edocecrous,

I discussed your issue with the team, it's serious enough that we want to try to resolve it. 
We've added a ticket to the backlog with the intent to improve this. 
For internal reference CURA-10872

Thanks for the report! 👍

Edocecrous, 2023-08-02:
[Plug_funnel.zip](https://github.com/Ultimaker/Cura/files/12238791/Plug_funnel.zip)
Just rotated it so supports are needed, and even on preview the support is printed first.
Cura 5.4.0, Linux

saumyaj3, 2023-08-03:
> [Plug_funnel.zip](https://github.com/Ultimaker/Cura/files/12238791/Plug_funnel.zip) Just rotated it so supports are needed, and even on preview the support is printed first. Cura 5.4.0, Linux
> 
> btw, The windows user who reported same problam was not me, i"m the original reporter. [image]

Hey! Thanks for another test case. I was asking to another user, so that I can have the issue recreated for another machine too. Anyway, this issue should be fixed in the later release. Thanks for your report :) 👍

closed 2023-08-09 by MariMakes