Ultimaker/Cura #15091: Strange print order for supports sticking out from "holes"

pietryna, 2023-03-30:
### Application Version

5.3.0

### Platform

Windows 11

### Printer

Creality Sermoon D1

### Reproduction steps

I've loadded my model and just sliced it. Did not pay much attention to exact nozzle path but when I've looked lated into the printer I've realized that initial support structure is somehow failed. (thankfully supports finally has been printed correctly so the supported structures should be OK)

So steps are 
1. Load model
2. Flip it 180 deg top/bottom
3. Slice with normal support.

### Actual results

Structures are pretty rectangular here
Result of the slicing is something like that:
[image]
[image]
Support layer was extruded midair without any support, then nozzle try to interface it with the model. This seems to be quite impossible. Even with high cooling (which is disabled for ABS print).
[image]

### Expected results

To me it looks like at least we should start with printing interface and then try to grid it with single line however I doubt it will be possible for non cooling materials like ABS, or without special bridging settings for this structure.

Best option in my opinion is to start support interface on top of model (not on side with layer before) and gradually overhang it with max 45 deg until we reach desired support structure base layer.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[model_and_gcode.zip](https://github.com/Ultimaker/Cura/files/11110867/model_and_gcode.zip)

rburema, 2023-04-13:
Are Support Z Distance or Support Bottom Distance options you could try out?

pietryna, 2023-04-14:
@rburema what do you mean?

pietryna, 2023-05-09:
@MariMakes 
Thanks for response :)
In fact it's hard to judge if it's bug or not. To me, any setting should not lead to air printing but I'm aware that might not be that easy thing to change/fix. I have no knowledge on how slicers are designed, especially at handling of such corner cases that might be counted in hundreds of thousands.

Anyway I was not aware that support horizontal expansion was causing supports to get out from model. It has changed when I've moved from 5.2.0 to 5.3.0. At this model this setting seems to be pointless, while when supporting through-holes it helps a bit with support removal.
Also I was not aware of this experimental feature to print supports in chunks. I was using support blockers to achieve that. Nice it's already in place :D 

Anyway, it's up to you to decide how to handle this issue.

MariMakes, 2023-05-10:
Hey @pietryna,

Thanks for the update 👍
In that case, I'll close the issue to be able to focus on the most requested and needed features and bug fixes from our community and from Ultimaker.

If you run into anything else, you are welcome to make a new ticket 😊 
Happy printing 🎉

closed 2023-05-10 by MariMakes