Ultimaker/Cura #15094: Wire printing slicing failed

Derox03, 2023-03-30:
### Application Version

5.3.0

### Platform

Windows 10

### Printer

Elegoo Neptune 3

### Reproduction steps

1 Select "Fine Quality" default profile
2 Enable wire printing
3 Try to slice a cube 50x50x50mm
[Cube.zip](https://github.com/Ultimaker/Cura/files/11114709/Cube.zip)

### Actual results

Cura stops in the middle of slicing and give me an error message: "Slicing failed with an unexpected error. Please consider reporting a bug on our issue tracker."

### Expected results

Slice my cube with wire printing enable

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

I also tried like 20 different models and none could slice

fieldOfView, 2023-03-30:
Unfortunately it looks like wireprinting is being removed entirely for the next release of Cura, so this will not be fixed.
https://github.com/Ultimaker/Cura/pull/15067

Having said that, perhaps this bug is similar to the flurry of "Slicing failed" errors being reported recently. Could you try moving the model to a (slightly) different location on the buildplate?

MariMakes, 2023-03-31:
Hey @Derox03,

Welcome to the UltiMaker Cura Github 🚀 
Thanks for your report 👍

@fieldOfView is correct, as mentioned in the release notes, it's broken in 5.3 and we are removing it because we see it's rarely used. So if you want to use it you'll have to go back to Cura 5.2 or older. You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.2.2

Can you help me understand why you wanted to use wire printing?

Derox03, 2023-04-03:
I was trying to make a cube shaped net and my previous results with normal printing were not really that good.  Anyway, I'm sorry to hear that wire printing has been removed, I consider it a very artistic and fast alternative to the standard printing method and even though I know that many use 3D printing to only make "solid" stuff having an option to have some fun was really nice.

MariMakes, 2023-04-04:
Hey @Derox03,

Thanks for getting back to us. 
A cube-shaped net sounds really cool. 

I'm also sorry to see it go but takes a lot of time to maintain and it often breaks.
But if we have enough testimonials I might be able to convince the team to bring it back. 
So more testimonials are very welcome. 

I will be closing the ticket because for now, it's not something we are going to fix, I hope you understand.

closed 2023-04-04 by MariMakes

ivanterzo, 2023-08-29:
> 
The wireframe function was interesting and useful for getting the results you can see in the picture

[image]