Ultimaker/Cura #15099: Slicing FAiled with Unexpected Error

tasif520, 2023-03-31:
### Application Version

5.3.0

### Platform

macOS Monetery

### Printer

Ultimaker 3

### Reproduction steps

Tried to slice a student project - just like I have many times before

### Actual results

[image]

### Expected results

should have sliced & given time for print

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[UM3_keychain-david Gheneim.3mf.zip](https://github.com/Ultimaker/Cura/files/11121781/UM3_keychain-david.Gheneim.3mf.zip)

MariMakes, 2023-04-05:
Hey @tasif520,

Welcome to the UltiMaker Cura Github 🚀 
Thanks for your report 👍

I can reproduce your crash, this is definitely a model that shouldn't crash. 
Luckily our most common workaround seems to work for this model. 
I can slice the model if I rotate it 90 degrees, that solves a number of slicing errors. 
[image]

I apologize for the inconvenience. 
I'll have someone from the team take a look at the slicing error.

jellespijker, 2024-01-14:
Thank you for your report. It seems that your uploaded project file slices correct on our latest unreleased 5.7.0-alpha.0+3abb5 which indicates that the issue you're experiencing seems to be fixed for the upcoming 5.7.0 release.
I'm closing this issue as fixed for the 5.7.0 release
[image]

closed 2024-01-14 by jellespijker