Ultimaker/Cura #15102: G-code flavor

julianvolden, 2023-03-31:
### Application Version

5.3.0

### Platform

windows 11

### Printer

Voron 2.4 350

### Reproduction steps

When configuring my freshly built Voron in cura it automaticly chooses reprap gcode flavor.

### Actual results

.trying to change it back to marlin does not help. as soon as you close the config/press any of the check boxes  it changes back to reprap flavor

### Expected results

It should have stayed at the preffered gcode flavor that is beeing choosen

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]

GregValiant, 2023-04-03:
Thanks for the report.
That does seem to be broken.  It appears that it isn't the only problem with the "Machine Settings" dialog.
@MariMakes might have some information about this.  I'm going to mark this as a duplicate of #15120 as it is the same dialog with the same behavior (a setting won't change).

julianvolden, 2023-04-03:
Thank you!

i found another way around it aswell, deleted cura 5.3 (and all included profiles/files for 5.3) 
installed cura 5.2, made a voron printer profile with Marlin flavor.  Then I installed Cura 5.3 again 

No it is Marlin gcode flavor

MariMakes, 2023-04-04:
Hey All.

It's a duplicate of https://github.com/Ultimaker/Cura/issues/14795
It's already been resolved and should be ready for an upcoming patch release (one that also includes the correct translations) 

Thanks for your report 👍

closed 2023-04-04 by MariMakes