Ultimaker/Cura #15103: Post-processing scripts do not work correctly when printing sequentially "individually"

gumileeev, 2023-03-31:
### Application Version

5.3.0

### Platform

win10 64

### Printer

FBG5

### Reproduction steps

1. I place several objects on the table
2. Activate sequential printing "individually"
3. I select the "ChangeAtZ 5.3.0(Experimental)" option in the post-processing script
4. I choose: 1. enable, 2. height trigger (or I even tried "layer No.", specifying either layer 12 to start and 95 to turn off the fan. I even tried setting the height for each shape separately, according to the preview, according to the total height of 1917 layers from 9 shapes: 1, = 1, = 12-95
2,=226-309
3,=439-522
4,=652-735
5,=865-948
6,=1078-1161
7, = 1291-1374
8, = 1504-1587
9, = 1717-1800)
3. I go down, select "Change Fan Speed", in one script I start at a height of 12 and fan 0 at a height of 95 (similarly, I try to turn it on and off at a height of 2.4 at a height of 19mm), or for each of the identical figures separately, as in the table above.

### Actual results

All this does not work at all, the fan is activated only for the first figure, and does not work for the rest of the figures

### Expected results

I expected that once the height in mm to activate and deactivate the fan would work for all shapes. By layers, I thought it would work the same way, or at least the way I described above, when the layers for each figure are described separately, according to the described height for each figure, according to the vertical preview line.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

**I see a big problem in the fact that the preview is not available for the fan. Please add fan operation to the "preview" menu. Without this, it is impossible to check the correctness of the cooling settings of the model in the right places :(**

GregValiant, 2023-04-01:
Thanks for the report.
Unfortunately, although there are 1917 layers in the preview, there aren't that many in the gcode.  What there is are 9 layer:0's, 9 layer:1's, 9 layer:2's, etc.  When ChangeAtZ runs it finds the first instance of the height (or Layer) and puts in the change, then it moves on.  It never gets to the next instance of the layer which would occur in the next part.

A workaround is to use a couple of the "Search and Replace" plugins as they work throughout a gcode file.
For example:
Search = ;LAYER:6\n
Replace= M106 S100\n;LAYER:6\n
Use Regular Expressions = Checked

That would result in a change at every "LAYER:6" in the file. 
M106 S100
;LAYER:6

The "\n" separator indicates a newline character.

gumileeev, 2023-04-02:
> I have a pull request in for this Post Processor.

Thank you! It works great! It works the way I wanted! I hope that it will appear in stable versions in the assembly. 
I hope that one day in Kura, in the preview, you will be able to see the speed of printing, the speed of the fans, as it is done, for example, in a prusa

GregValiant, 2023-04-03:
The "Cooling" section of Cura will turn the fan on, but once it's on, that's it.  So the preview would show the fan off at the beginning and then at the max speed (as set by the user) at some layer, and then it would stay that way for the rest of the print.  So given the current situation it would be pretty pointless to go through all the work of coding to show that.

If you open a "One at a Time" gcode in Cura you will see that it doesn't preview correctly.  That's another problem.  If the layers were re-numbered from 0 to 1917 then you could have Pauses or Fan Changes at different points in the file.  I'm working on that.  The reason there are repetitious layer sections is that "Initial Layers" need to be handled differently by the Cura Engine.  The printer doesn't care though so it appears the layers could be renumbered with no ill effects in post-processing prior to running a Fan or Pause post-processor.  That needs more looking into though.

At any rate, the way it works right now isn't a bug but rather the dreaded "expected behavior".  I'll go ahead and close this since if the "Advance Cooling Fan Control" post processor is accepted that would just leave the Cooling Preview as a problem and that would need to be handled via a feature request.  I hope you understand.

closed 2023-04-03 by GregValiant

GregValiant, 2023-04-03:
I need a tester again.  Since you print some models in One at a Time mode I thought you could make use of this.
Here is another post processor for "One at a Time".  It re-numbers the layers.  You run this one first, and then you can use "Pause at Height" or "Filament Change" or "Advanced Cooling" at a layer as shown in the Cura preview.  You could print different models different colors if you wanted to.
A nice side effect is that  a gcode that was made with this post processor running will show in the Cura preview exactly the same as the original preview did.
This doesn't have any settings.  It simply renumbers all the layers in consecutive order.  It does need to be the first Post that runs so something like Pause at Layer running behind it can work with the new layer numbers.

[Renumber_OneAtATime_Layers.zip](https://github.com/Ultimaker/Cura/files/11143071/Renumber_OneAtATime_Layers.zip)

gumileeev, 2023-05-01:
> I have a pull request in for this Post Processor. If you don't mind being a Guinea Pig then unzip the file and copy "AddCoolingProfile" into the Scripts folder in Cura (C:\Program Files\UltiMaker Cura 5.3.0\share\cura\plugins\PostProcessingPlugin\scripts).

I found the problem. If I set the fan % from 1 to 12% - the fan speed will be 12%. Maybe this problem has already been fixed? Where can I download the updated version?

GregValiant, 2023-05-02:
If you are talking about my post processor Advance Cooling then yes, the minimum fan speed is 12%.  There aren't very many fans (almost none) that will start to spin at speeds below that.  The norm for fan start is around 20% duty cycle.

gumileeev, 2023-05-02:
> If you are talking about my post processor Advance Cooling then yes, the minimum fan speed is 12%. There aren't very many fans (almost none) that will start to spin at speeds below that. The norm for fan start is around 20% duty cycle.

isn't it better to control the minimum fan value on the printer side? Can I somehow change the value myself? 12 is a lot for me, I think 8 would be more suitable for me. Or it would be great if users had a choice, a checkbox, with the ability to remove or change this restriction. Kura, on the other hand, does not worry about what minimum the fan can show. I really hope that you show the solution for me, I'm using it right now.

GregValiant, 2023-05-02:
There are "12%" comments in 
Line 33, 775, 758, 788, 792
These lines are the logic/math
Line  776, 777. 793(2)
You can open the file in any text editor and make the changes.

I had been thinking about a "Start Up Speed" but the post-processor is already as long as War and Peace so I passed on it and just used a minimum speed.  It's based on the slowest my cooling fan will spin.  I talked to a couple of Ultimaker owners and their minimum is about 20%.  You must have a heck of a cooling system.  The difference between 8% and 12% on my fan is about 0.4CFM or 0.01mm³/min.