Ultimaker/Cura #15105: No Temperature added to g-codes

markparra26, 2023-03-31:
### Application Version

5.3.0

### Platform

Windows 10

### Printer

Creality Ender-3 S1 Pro

### Reproduction steps

Sliced and STL
Added to my printer.
Temperature was 0 on Nozzle

### Actual results

Print never started

### Expected results

Stl should print

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[CE3S1PRO_Screw_support.zip](https://github.com/Ultimaker/Cura/files/11125541/CE3S1PRO_Screw_support.zip)

GregValiant, 2023-04-01:
Thanks for the report and the file.
I can't reproduce this.  I sliced your project file and at the beginning of the gcode is:

;Generated with Cura_SteamEngine 5.3.0
M140 S60
M105
M190 S60
M104 S200
M105
M109 S200
M82 ;absolute extrusion mode
; Ender 3 S1 Pro Start G-code

The M104 and M109 lines are the hot end temperature and the M140 and M190 lines are the bed temperature.
If this was a Cura bug nobody would use the software.  I think you have a local problem.  If you open the gcode file in a text editor you should see the lines noted above.
I'm going to remove the bug label and close this.  Respond here if you feel that is premature as it can always be re-opened.

closed 2023-04-01 by GregValiant

markparra26, 2023-04-05:
Hello,

I have installed Cura 5.3.0 in another computer. Ive selected printer
Creality Ender 3 S1 Pro. When I slice Slt in this version and try to print
them, i get no temperature on the nozzle.
I don't know if this is a bug when selecting this printer.

Regards,