Ultimaker/Cura #15106: Multiple Nozzle Settings for Creality Ender 3 S1 Pro not available

Geoloog, 2023-03-31:
### Application Version

5.3.0

### Platform

Windows 10

### Printer

Creality Ender 3 S1 Pro

### Reproduction steps

1. I tried to change the nozzle size for the printer from 0.4mm to 0.6mm but it is not availble for this printer

### Actual results

1.  Nozle size cannot be changed

### Expected results

In Cura 5.2.2 it is possible to choose larger and smaller nozzle sizes than 0.4mm

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

No other comments

GregValiant, 2023-04-01:
Thanks for the report.
The nozzle sizes come from the Creality Printer Definition files which were submitted to Ultimaker for inclusion in Cura.  The S1 Pro only has a .4 nozzle available but the regular S1 has a full set off nozzle sizes.  You should put in a ticket with Creality since it is their definition that seems to be the problem.
In the meantime you could install a regular S1 and all the nozzles would be available.
I will leave the bug label on for the Cura team but this does appear to be a Creality problem.

Geoloog, 2023-04-02:
Thank you for your very prompt reply. I understand what you are saying.

Be Blessed.

Hendrik Bosman
South Africa

GregValiant, 2023-04-02:
This is for Windows but Mac and Linux would be similar.
Go to the "C:\Program Files\UltiMaker Cura 5.3.0\share\cura\resources\**variants**\creality" folder.
You will see that the only nozzle for the S1Pro is "creality_ender3s1pro_0.4.inst.cfg" while right above is the 1.0 nozzle for the S1Plus model "creality_ender3s1plus_1.0.inst.cfg" file.

Select all the nozzle files for the S1Plus and "Copy" them to the configuration folder --
"C:\Users\Valued Customer\AppData\Roaming\cura\5.3\**variants**"

Rename each file from "creality_ender3s1**plus**_X.X.inst.cfg" to "creality_ender3s1**pro**_X.X.inst.cfg"
Open each file in turn and change the line:
[general]
definition = creality_ender3s1**plus**
to
definition = creality_ender3s1**pro**
Restart Cura and you should be good to go.
Moving the files to your Configuration Folder will insure that when you upgrade Cura that the nozzles will still be available in the new version.

I'll go ahead and close this since the author of the S1pro definition forgot to include the other nozzle files.

closed 2023-04-02 by GregValiant

motionbug, 2023-04-11:
@GregValiant  thanks, that fixed my problem