Ultimaker/Cura #15110: Brim Only On Outside not working

SHVines, 2023-04-01:
### Application Version

5.3.0

### Platform

Windows 11

### Printer

Artillery Sidewinder X1

### Reproduction steps

Set:
"Brim Only on Outside" setting
"Bottom layers" set to 0
"Brim Distance = 0"

[artillery_sidewinder_x1_brimonlyonoutside.inst.zip](https://github.com/Ultimaker/Cura/files/11129349/artillery_sidewinder_x1_brimonlyonoutside.inst.zip)

### Actual results

Filled my 1mm hole on bottom layer in

### Expected results

Should have had a 1mm hole on bottom layer.
To work around, set "Brim Distance" to 0.4m

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[Fus5.zip](https://github.com/Ultimaker/Cura/files/11129350/Fus5.zip)
[image]

GregValiant, 2023-04-01:
Thanks for the report.
Without a project file it's impossible to tell what's going on.  Are those blind holes?  If so then maybe that isn't a brim but rather supports for the top of the hole?
Use the "File | Save Project" command to create a 3mf project file.  Zip that file and post the zip folder here.

SHVines, 2023-04-01:
Hi,

Sorry new to Cura and raising support calls. Please find the Project file
attached.

The issue is the bottom layer of the hole that runs down through the
project.

Best regards,
Simon

GregValiant, 2023-04-01:
You need to actually post the zip folder here.  Using a phone doesn't work.  All that comes through is a link to this post.

SHVines, 2023-04-01:
Sorry thought responding to email would work. Here it is:

[ASX1_Elev1L.zip](https://github.com/Ultimaker/Cura/files/11130306/ASX1_Elev1L.zip)

GregValiant, 2023-04-02:
Got it.  Thanks.
Here is the problem.  The hole in the model you provided is not a "hole" but a slot that goes to the outside of the model.  Here is a screenshot looking up through the bottom of the build plate.
[image]
So this isn't a Cura bug but rather an unfortunate feature of the model.  That little bit of brim _IS_ on the "outside" because there is no "inside" there.
I'm going to remove the bug label from this.  I hope you understand.

Just talking printing now, that is one tough model to print.  The preview doesn't look good at all.
Here is the "foot" feature with your settings.  You can see that  the top layer will just be strings as they aren't in contact with the layers below.
[image]

Here is the "Fus5" model.  You can see that the small holes are actually keyhole slots.  It's the same problem in that there is no "inside".
[image]

SHVines, 2023-04-02:
Thanks for your very fast support and the explanation. It makes sense.
No problem closing the issue.

Really liking the software, pretty straightforward to use, given the number
of options available. Glad I chose Cura.

Simon

GregValiant, 2023-04-02:
I looked at them again this morning.  It appears they may have been designed to use "Sprialize Outer Counter".  Horizontal areas never print well with spiralize and so those areas will need post-printing attention and maybe some body filler, but the previews look better (other than that "Foot" which continues to be a problem).
I'll go ahead and close this.  Good luck with those.

closed 2023-04-02 by GregValiant

SHVines, 2023-04-05:
Thanks very much for your help. I will have a play with Spiralize Outer
Contour.

Simon