Ultimaker/Cura #15111: 5,3 support problem

skamlam, 2023-04-01:
### Application Version

5,3

### Platform

windows 11

### Printer

any printer

### Reproduction steps

try making a support test with the calibration parts plugin using normal supports everywhere
supports will trap the model inside making it impossible to remove even if they can be broken off

### Actual results

supports will trap the model inside making it impossible to remove even if they can be broken off which did not happen with the same settings in any previous cura

### Expected results

the supports should not trap the model inside

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]

skamlam, 2023-04-05:
> Hey @skamlam <https://github.com/skamlam>,
>
> Welcome to the UltiMaker Cura Github 🚀
> Thanks for your report 👍
>
> In Cura 5.3 the default Support Horizontal Expansion has increased.
> You can find a mention of this in the release notes:
> *Updated Support Horizontal Expansion, Support Join Distance, and Minimum
> Support Area Settings to improve support reliability.*
>
> Can you reduce the Support Horizontal Expansion, for me and let me know if
> that resolves your issue?
> Because to me, it doesn't seem like a bug in Ultimaker Cura.

Thank you very much ! I totally missed that setting
Fixed the problem.

MariMakes, 2023-04-06:
Great! Good to hear your issue has been resolved.
I'll close the issue. Happy printing! 💪

closed 2023-04-06 by MariMakes