Ultimaker/Cura #15113: Slicing Failed

theOhmbrewer, 2023-04-01:
### Application Version

5.2.1

### Platform

Windows 11

### Printer

Ender 3 Pro

### Reproduction steps

1. Added 8 individual pieces positioned closely but not touching.
2. Modified printing settings to meet the needs of the detail I am wanting
3. 0.12mm nozzle and no settings out of the ordinary
4. Start slicing
5. About 1/8th in, it pauses and fails

### Actual results

Slicing failure - Slicing failed with an unexpected erorr.

### Expected results

Slike my print and export stl

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[CE3E3V2_Wall-Group.zip](https://github.com/Ultimaker/Cura/files/11129970/CE3E3V2_Wall-Group.zip)

theOhmbrewer, 2023-04-01:
[cura.log.zip](https://github.com/Ultimaker/Cura/files/11130003/cura.log.zip)

MariMakes, 2023-04-05:
Hey @theOhmbrewer,

Welcome to the UltiMaker Cura Github 🚀 
Thanks for your report 👍

The new Cura 5.3 contains the fix for this for a number of slicing crashes. 
I was able to slice your model successfully in 5.3 Stable.
[image]

Could you test if that's the case for you as well? 
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.3.0

If you still run into a crash, please double-check if you shared the correct file and let us know.
I'll take another look at what is going wrong. 👍

closed 2023-04-20 by github-actions[bot]