Ultimaker/Cura #15114: Config Error

WigandClone, 2023-04-01:
### Application Version

5.3.0

### Platform

Windows 10

### Printer

SV06

### Reproduction steps

1. Update from 5.2.1 to 5.3.0
2. Launching 5.3.0
3. Getting "Konfigurationsfehler"
[image]

4. Not understanding why 5.2.1 is able to read the files and 5.3.0 is not.

### Actual results

Nothing happened. I closed the programm and used 5.2.1 instead.

### Expected results

Launching the program without Error-messaging

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/11130113/cura.log)

WigandClone, 2023-04-05:
As far as I understand as a beginner....

I have loaded a **project** from 5.2.1 into 5.3.
Instead of the originally set temperature of 220°C, 
it now says 200°C in the material settings. 
So I have to assume that the old data was not converted correctly. 

And if I understand the optimizer correctly, it just skips the incompatible profiles.
It does not try to convert them. Or am I wrong?

MariMakes, 2023-04-06:
Hey @WigandClone,

Cura will always listen to the value of the setting that is displayed in the Print Settings on the right side of your screen. 

So in this picture:
[image]

The printjob will have a printing temperature of 220 instead of the default printing temperature of 200.
The Blue arrow indicates that it's a setting that you changed yourself.
The Fx indicates that this setting is normally calculated but it currently has an absolute value set.

Right now Cura sees that in previous versions of Cura you've had Sovol printers loaded and remembered the settings you changed manually back then. If you install the Start-Up Optimizer Plug-in, it will ask if you would like to hang on to the old configurations and will tell Cura to not load them anymore so you don't get an error anymore.

I don't think you will lose any settings, but you can make a back-up just in case you don't like it and want to return to a state before you installed the plug-in. 

Does that make sense? 
Can you install the plug-in and let us know if that resolves your issue?

WigandClone, 2023-04-09:
Sorry, but the various influences of printer or material or profiles etc. confuse me more than they help me. I don't see a clear structure in the procedure here. 
As for the plugin, all I can say is that it does not convert what it thinks are the wrong profiles, but skips them. Unfortunately, neither Cura nor the plugin tells me what they are interfering with. After all, the profiles were created by me on the old Cura. On the old Cura these profiles had obviously made no problems. How should I as a beginner now get out where the problem lies. In principle, I can now only start to try out all the profiles again. This is too much work and time consuming for me. I would have to retest all materials, because I have saved them as profiles of the printer, and that only on the advice of other users. I myself have been programming on different systems for about 30 years and I think that a new version of a program should be able to read and convert the previous file formats (and without needing an extra plugin that just **skips** the "faulty" profiles.) And without the user having to fear losing a setting. Of course it is not the same the other way around. An old version does not have to be able to read the file format of the new version.

MariMakes, 2023-05-04:
Hey @WigandClone,

Working with profiles is indeed confusing. It's grown quite bulky over the last few years and it's something we want to improve for beginners. 

If you have experience with programming perhaps this page on our wiki explains our setting structure in more detail. 
https://github.com/Ultimaker/Cura/wiki/Getting-A-Setting-Value

I'm having some trouble seeing if your issue has been resolved. 
Would you still like to keep this issue open or do you agree that we can close it?

closed 2023-05-30 by github-actions[bot]