Ultimaker/Cura #15115: Build Plate Temperature Settings are gone. I can´t set a build plate temperature anymore... Not on the 5.2.2 and 5.3.0 Version. What should I do?

Titanelder, 2023-04-01:
### Application Version

5.3.0

### Platform

Windows 10

### Printer

Creality Ender 5 Plus

### Reproduction steps

1. Search for Materials
2. Try to set build plate temperature
3. Can´t find the setting (I´m searching, not on the control panel etc. won´t work)

### Actual results

I can´t find build palte temperature settings

### Expected results

I can set build plate temperature for PLA for example

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Can you help me?

motoyasu-atsumi, 2023-04-03:
I have same issue.

The reproduction method is as follows
1. change the Build plate shape from Rectangular to Elliptic in the printer settings.
2. change the Build plate shape in the printer settings back to Elliptic.
3. Build Plate Temperature will no displayed.

motoyasu-atsumi, 2023-04-03:
Sorry, Heated Bed was not checked. When I checked it, it was OK.

rburema, 2023-04-12:
@Titanelder Indeed, check if heated bed is on in the machine settings (Manager printers -> Machine settings) If this is indeed already on, check the settings visibility menu (in the custom settings menu, next to the search bar, click the hamburger menu, select 'all' if that doesn't work, you can diagnose the problem with 'Manage Settings Visibility')

MariMakes, 2023-05-04:
Hey @motoyasu-atsumi,

Glad to see your issue is resolved. 
To me, it doesn't seem like a bug in Ultimaker Cura anymore. 
I'll be closing this issue. I hope you understand.

closed 2023-05-04 by MariMakes