Ultimaker/Cura #15119: Printer definition lowered zhop speed, not savable in profiles.

helifino, 2023-04-01:
### Application Version

All versions tested (4.2+)

### Platform

win10

### Printer

Geeetech A30

### Reproduction steps

Reproduce:
 - Try to slice with a default slicing profile, 'Enable z-hop' unchecked
 - Get error 'z-hop speed too high'
 - Set z-hop speed from default 10 to 7, even though z-hop is unchecked.
 - Try to save as new slicing profile, option is greyed out.

### Actual results

'z-hop speed' is retained when slicing with the same profile but when switching to another it reverts.

### Expected results

Either the 'Save' option should enable, or (preferably) the printer max speeds should override the set speed if too high. Users shouldn't have to remember to set an obscure unused option every time they switch between multiple printers and profiles.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Unnecessary section, intentionally left not blank

GregValiant, 2023-04-03:
Thanks for the report.
The definition file for the Geeetech A30 (and maybe other Geeetech definitions) lists the Max Z speed as 7mm/sec.  They did not put in an over-ride for the "Zhop speed" setting in Cura and so yes, they are in conflict.  The definition file should contain a line:  "speed_z_hop": { "default_value": 7 } to insure that the setting was not above the Max of the printer.  That part is on either Geeetech or the authors of the file ("William & Cataldo URSO") as it is their definition file.
I didn't have any trouble making a new Settings Profile after I changed the speed to "7" and so I can't duplicate that part of the problem.
If you load the "Printer Settings" plugin from the Market Place and then restart Cura - Printer Settings will load as the last setting category in the list.  Within Printer Settings is "Maximum Speed Z" and you can set that to "10".  It will not have an effect on the printer, but you would be able to put "10" into the Zhop Speed box in Cura.  Maybe that will change the ability to make a new profile based on your current profile.

I'll leave the bug label on this for the Cura team but it appears to be a problem with the definition file that was provided by others and submitted to UltiMaker for inclusion in Cura.  UltiMaker has no idea what settings might be appropriate for other manufacturers printers.  I hope you understand.

helifino, 2023-04-04:
I shouldn't have to make a new custom profile, it should limit it automatically without intervention (at most, with a warning popup) or allow saving the setting back to the default profiles (perhaps as a "local override" like machine profiles).