Ultimaker/Cura #15120: Cannot set bed shape to elliptic

Matszwe02, 2023-04-01:
### Application Version

5.3.0

### Platform

Windows 10

### Printer

Custom FFF Printer

### Reproduction steps

1. Create a custom FFF printer
2. try to change bed shape to elliptical
3. after closing printer menu and opening again, it's back to rectangular

### Actual results

not working

### Expected results

working

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

-

Matszwe02, 2023-04-01:
The elliptic shape can be achieved by adding the "machine_shape = elliptic" line in definition_changes/settings.inst.cfg file, doing it by UI doesn't work.

GregValiant, 2023-04-03:
Thanks for the report.
I can confirm that it worked in 4.13.1 but does not in 5.3.0.  (I didn't check other releases.)
Someone from the Cura team will take a look.

Matszwe02, 2023-04-03:
It worked in 5.2.0 (or .2), that's how I figured out how to change it manually. So that's the issue with the latest release

MariMakes, 2023-04-04:
Hey All.

It's a duplicate of https://github.com/Ultimaker/Cura/issues/14795
It's already been resolved and should be ready for an upcoming patch release (one that also includes the correct translations) 

Thanks for your report 👍