Ultimaker/Cura #15123: slicing failed

BenHuset, 2023-04-02:
### Application Version

5.2.1

### Platform

windows 10

### Printer

creality CR-10S

### Reproduction steps

import file HLS_Starship.stl
rotate model so landing gear  is up
change Z to -328.62mm   to print bottom of model as it's too big to print in one piece.
Click Slice button

[$$-Human_Landing_System__HLS__5470360.zip](https://github.com/Ultimaker/Cura/files/11131185/-Human_Landing_System__HLS__5470360.zip)
[image]
[image]
[image]

### Actual results

slicing failed
unexpected error   (screen capture #3)
Get told report bug.    no err # or code given

### Expected results

gcode file created.
[cura.log](https://github.com/Ultimaker/Cura/files/11131187/cura.log)

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

none

MariMakes, 2023-04-05:
Hey @BenHuset,

Welcome to the UltiMaker Cura Github 🚀 
Thanks for your report 👍

The new Cura 5.3 Stable contains the fix for this for a number of slicing crashes. 
Could you test if that's the case for your model as well? 
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.3.0

Have you tried reverting to the default settings?
Have you tried moving the model away from the center of the buildplate and rotating by 90 degrees?

If that's not it, I'll need a little bit more information from you.
I tried to load your project file, but my Cura is unable to open it, it says it's a corrupt file. 
Do you mind trying it again?  To save a project file go to File -> Save project. Please make sure to .zip your project file. For big files, you may need to use WeTransfer or similar file-sharing sites.

You'll need to open Github in your browser to attach any files to your ticket
You can click the **view it on GitHub** link at the bottom of your email to open GitHub in the browser.

BenHuset, 2023-04-06:
MariMakes :

ok upgrading to ver 5.3.0 seems to have fixed this slicing problem.
thanks.

If you are interested the HLS_Starship.stl file came from
https://www.thingiverse.com/thing:5470360

--Ben

closed 2023-04-06 by BenHuset