Ultimaker/Cura #15124: The fill density selection option is blocked

Eddy71, 2023-04-02:
### Application Version

5.3.0

### Platform

Цштвщцы 10 64 bit

### Printer

DIY Prusa Mk3

### Reproduction steps

Fill density option is blocked and it is impossible to change the model fill value when printing

### Actual results

I can't change the fill density option. The choice field is not active

### Expected results

Nothing. Installed the program, but the field is not active
[image]

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

The problem started on version 5.2.x

GregValiant, 2023-04-02:
Thanks for the report.
When you set the "Infill Line Distance" manually then the setting becomes an over-ride and is un-linked from the parent Density % box.  Clicking on the Fx button next to the "Infill Line Distance" box will re-link the "child" setting Line Distance to the parent "Density%".
Let us know if that works for you.
Right now your Density box shows 20% but a Line Distance of 0.45.  At 0.45 Line Width you would get an Infill Density of 100%.

Eddy71, 2023-04-02:
> When you set the "Infill Line Distance" manually then the setting becomes an over-ride and is un-linked from the parent Density % box. Clicking on the Fx button next to the "Infill Line Distance" [...]

Thank you, bringing the line spacing field back to default has revitalized the fill density entry field

-- 
С уважением, Эдуард.
-----------------------------------
http://www.eddy.com.ua/

https://www.facebook.com/eddy.com.ua

GregValiant, 2023-04-03:
Good.  That's how it's supposed to work.  I'll go ahead and close this then.

closed 2023-04-03 by GregValiant