Ultimaker/Cura #15126: Slicing Failed

bricomp, 2023-04-02:
### Application Version

5.3.0

### Platform

Windows 10

### Printer

Elegoo Neptune 3 Plus

### Reproduction steps

Dropped item ont build plate (in Cura).
Pressed Slice ... It failed after a prolonged period of time
Cura 5.3.0 Alpha + Xmas worked just fine

### Actual results

Cura reported failed, suggested sendin error report

### Expected results

Slicer file for printer expected

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[EN3P_hhm frame11a.zip](https://github.com/Ultimaker/Cura/files/11132772/EN3P_hhm.frame11a.zip)
[cura.log](https://github.com/Ultimaker/Cura/files/11132776/cura.log)

MariMakes, 2023-04-05:
Hey @bricomp,

Welcome to the UltiMaker Cura Github 🚀 
Thanks for your report 👍

I took a look at your file and was a bit surprised about the orientation. 
If you turn your model around like this, it doesn't crash, it's easier to slice, and uses material.
[image]

I also noticed that the model slices as expected when you chose normal support instead of tree support. 
[image]

To me, it doesn't seem like a bug in Ultimaker Cura anymore, or can you share why it's important to print your model like this and with tree support instead of regular support? 

Would you still like to keep this issue open or do you agree that we can close it?

bricomp, 2023-04-05:
I don’t see why orientation should matter since the bed is square, i.e. the same size on width and depth.

But OK go and close it.

Tree supports are used because they come off so much easier and use less material.

 

Regards

Bob Bridges

 

 <https://user-images.githubusercontent.com/40423138/230148736-1d5a1d38-64e7-417d-91fc-a975231c6680.png> 

 <https://user-images.githubusercontent.com/40423138/230149199-fac54954-aa93-45d8-8996-837a30fe846f.png>

MariMakes, 2023-04-06:
Thanks for the update 👍 
I'll close this issue. 

Happy printing!

closed 2023-04-06 by MariMakes