Ultimaker/Cura #15127: Wire printing "failed"

salamanders, 2023-04-02:
### Application Version

5.3.0

### Platform

OS X

### Printer

Prusa Mk3s

### Reproduction steps

Load up a 3mb from openscad
Change 1 setting: "Wire Printing"
Slice

### Actual results

"Slicing failed with an unexpected error"

### Expected results

Slices model.
[PI3MK3M_hilbertcoaster-v2.3mf.zip](https://github.com/Ultimaker/Cura/files/11133019/PI3MK3M_hilbertcoaster-v2.3mf.zip)
[cura.log](https://github.com/Ultimaker/Cura/files/11133020/cura.log)

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

Just started using Cura.  Changed nozzle size to 0.6.

MariMakes, 2023-04-03:
Hey @salamanders,

Welcome to the UltiMaker Cura Github 🚀 
Thanks for your report 👍

As mentioned in the release notes, it's broken in 5.3 and we are removing it because we see it's rarely used. So if you want to use it you'll have to go back to Cura 5.2 or older. You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.2.2

Can you help me understand why you wanted to use wire printing?

salamanders, 2023-04-03:
Cool, thank you for the reply!

It was an experiment: I wanted a meshy-like wall to hold desiccant beads in a container, so I could keep my PET-G dry.  I bet I can get by with perimeters = 0 and using the right infill.  Or maybe drilling holes in the container post-processing.

MariMakes, 2023-04-04:
Nice! That sound like a cool application ❤️