Ultimaker/Cura #15128: Plugin: Sidebar GUI makes cura unusably slow

Slopecarver, 2023-04-02:
### Application Version

5.3

### Platform

windows 11

### Printer

ender 3 pro

### Reproduction steps

Install Sidebar GUI plugin
Restart
Load Cura

### Actual results

Sidebar GUI caused cura to take over 1 minute to load and every click another minute in the menu just to turn off the plugin.

### Expected results

no molasses

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

N/A

GregValiant, 2023-04-03:
Thanks for the report.
I'm also running Windows 11 with an Ender 3 Pro and I haven't had any issues with the SideBar GUI.
@fieldOfView developed it and he's very good about keeping abreast of issues with his plugins.  You can go to his [GitHub Page](https://github.com/fieldOfView/Cura-SidebarGUIPlugin/issues) and fill out a ticket there.  There may be some more information regarding performance issues as well.
He may need to know more about your computer set up as well (how many monitors, what video card or integrated video system, those sorts of things).

fieldOfView, 2023-04-03:
Specifically, see this issue:
https://github.com/fieldOfView/Cura-SidebarGUIPlugin/issues/29
(and this comment in particular: https://github.com/fieldOfView/Cura-SidebarGUIPlugin/issues/29#issuecomment-1413599582)

closed 2023-04-03 by fieldOfView

fieldOfView, 2023-04-04:
A "nightly" version of Cura seems to fix this problem. For more info, see https://github.com/fieldOfView/Cura-SidebarGUIPlugin/issues/29#issuecomment-1495981891