Ultimaker/Cura #15132: Creality Ender-3 "not connected"

baznr, 2023-04-03:
### Application Version

5.3.0

### Platform

Windows 10

### Printer

Creality Ender-3

### Reproduction steps

The printer functions correctly with Cura v.4.11.0.
But Cura v.5.3.0 can't see the printer

Probable very short connection time-out period?

### Actual results

Cura v.5.3.0 can't see the printer

### Expected results

Cura v.5.3.0 to see the connected printer.

It is so difficult to add a "connect printer" button and a setup for connections specs?

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Same setup. No cables moved.
[image]

GregValiant, 2023-04-03:
Thanks for the report.
Because no UltiMaker printers use USB printing the UltiMaker Company isn't going to pay their people to fix USB printing.  It is up to the community to do something about it.  You will notice that no one has picked up the ball on that.  There are so many combinations of hardware, software, and drivers out there that it's really not possible to get it to work for every combination.
My printer is an Ender 3 Pro and I haven't had any trouble connecting it to Cura on either Windows 7, 10, or 11.  I have my own app that I control the printer with over the USB and Cura often gets in the way (until I remember to turn off USB Printing in the Marketplace).
So if USB Printing works then it works and if it doesn't, then it doesn't.  You can try checking the port drivers and see if both versions of Cura are using the same port and driver combination.
Your other options for a print server would be Octoprint, Repetier Host, or PrintRun.

I'll leave the bug label on this so people see it.  Don't get your hopes up though.

baznr, 2023-04-04:
Thank you GregValiant!
Do you know in which file I can change the time-out connection delay?
In v.4.11.0 it was the file 
Ultimaker Cura 4.11.0\plugins\USBPrinting\AutoDetectBaudJob.py
but v.5.3.0 has changed that.

GregValiant, 2023-04-05:
I looks like there has only been a minor change to USB Printing in the last 3 years.  I believe that addressed a problem during printing and not with establishing a connection.  I won't comment on where a "time out" might be as I just don't know enough about it.
Have you tried slicing a file and creating the gcode using 5.3.0 and then opening that gcode file in 4.11 and sending it from there?  I realize it's a clunky workaround but it might get you going.