Ultimaker/Cura #15135: code 3221225477 /  Failing to discretize parabola

SirMaus, 2023-04-03:
### Application Version

5.3.0

### Platform

Windows 11

### Printer

Anycubic Viper

### Reproduction steps

I have sliced the model as always. Just like any other model. the error occurred with the same model already in 5.2, but "earlier" in the loading bar. Unfortunately i don't have any logs of that anymore...
The error also occurs with another stl, which is from another designer.
Other files from the same artists work fine.

Model and logs are attached. Model is free.

### Actual results

Slicing aborts with the message "unable to slice".
For more information the logs are attached

### Expected results

should slice properly

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[Cura Logs.txt](https://github.com/Ultimaker/Cura/files/11142053/Cura.Logs.txt)

Project download: https://cloud.bulldogicgames.com/s/iJ4SoQb7yGy2bFX

SirMaus, 2023-04-04:
Update:
If i change the scale to 99% or lower the slicing runs successful

rburema, 2023-04-13:
Thanks, I've added this issue to our internal ticket CURA-9558, which seems to be related.

Other workarounds that work are rotating the object slightly, instead of scaling.