Ultimaker/Cura #15136: Poor layouts on modal screens

N6REJ, 2023-04-03:
### Application Version

5.3.0

### Platform

win 10x64 pro

### Printer

Voxelab Aquila S2

### Reproduction steps

open a modal window in non full-screen mode.
you'll get a partial layout or poorly laid out window.
change to full screen reopen modal and it will be fine.
THIS IS NOT TRUE for colors.  They are always messed up.

### Actual results

window at roughly 50% 
[image]
after going full screen
[image]
[image]

### Expected results

[image]

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

sometimes the color modal is so messed up you can't even select all the colors

rburema, 2023-04-12:
We suspect several issues may be solved with our update of Qt to 6.4 (the library we use to make the GUI). If you whish, you could try if this alpha build solves your issues (warning for bugs, please keep your older Cura around when installing): https://github.com/Ultimaker/Cura/discussions/15098

N6REJ, 2023-04-13:
@rburema yes it does solve the windows issues.

closed 2023-04-13 by N6REJ