Ultimaker/Cura #15137: Additional "Use Existing" printer profile when openning a project

DECO74, 2023-04-03:
### Is your feature request related to a problem?

I have a saved Cura project (.3mf). When I open the project it asks me if I want to update the existing profile or create a duplicate one. It doesn't give me an option to use the existing profile. Instead it gives me the option of updating an existing profile with who nows what are the configuration changes that maybe I adjusted after and the existing one is a better suited profile; or it creates a junk profile that later I have to cleanup the mess ...

### Describe the solution you'd like

The easiest solution is to give a third-option for using the existing profile with the same name.

### Describe alternatives you've considered

A more complete solution would be an interface for comparing profiles and choosing to (update the/create a) profile.

### Affected users and/or printers

Everyone would benefit with this change as it gives a more appropriate handling to profiles when opening a project file.

MariMakes, 2023-04-04:
Hey @DECO74,

Welcome to the UltiMaker Cura Github 🚀 
Thanks for your report 👍

Could what you are describing be similar to the discussion mentioned here: https://github.com/Ultimaker/Cura/issues/14305
We added a ticket to our backlog to Include option to "ignore profile" when loading old project files.
Would that solve your problem as well?

DECO74, 2023-04-04:
Hi @MariMakes, 

Yes, the expected outcome of my request is the same as @GoGuyT3D report.

But I can delete these unwanted profiles without any crashes of the application.

Thank you! Have a fantastic day...

MariMakes, 2023-04-05:
Thanks for confirming!

The request is present in our backlog.
For internal reference CURA-10258

Thanks for the report! 👍