Ultimaker/Cura #15140: Can't update  "G-code flavor"

mattdog01, 2023-04-04:
### Application Version

5.3.0

### Platform

Windows 10

### Printer

custom

### Reproduction steps

I tried to change my printers G-code flavour but it won't save the change.  (I have updated the OS from Marlin to RepRapFirmware)  This was reported in 5.3.0 Beta 2 and is still here in 5.3.0 release see error report #14795

1) opened tab "Preferences"
2) opened "Configure Cura"
3) selected "Printers"
4) selected my printer "Matt's Hypercube"
5) clicked button "Machine Settings"
6) clicked on down arrow of the "G-code flavor" to see the drop down list
7) clicked on "RepRap" (it does show in the box after the list closes
8) Close "Machine Settings" by clicking on the "X"
9) Close "Preferences" by clicking on the "X"
repeat the process
10) opened tab "Preferences"
11) opened "Configure Cura"
12) selected "Printers"
13) selected my printer "Matt's Hypercube"
14) clicked button "Machine Settings"
15) looked at "G-code flavor" and it says "Marlin"
16) sob into my hands

### Actual results

No change

### Expected results

The change should be saved after closing the popup window

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

none to add at this point
[error report.zip](https://github.com/Ultimaker/Cura/files/11150256/error.report.zip)

GregValiant, 2023-04-04:
Thanks for the report.
The Machine Settings dialog continues to have problems since the move to QT6.  I'll mark this as a duplicate.  On one report the OP was able to fix the problem by deleting everything regarding 5.3.0 (including the configuration folder) and re-installing Cura.  #15102 also appears to be a duplicate.

MariMakes, 2023-04-05:
Hey @mattdog01,

Welcome to the UltiMaker Cura Github 🚀
Thanks for your report 👍

This seems to be a duplicate of https://github.com/Ultimaker/Cura/issues/14795
This has been resolved and the fix will be available in an upcoming patch release with the correct translations.

I'll be closing this issue as a duplicate but you can follow the progress here: https://github.com/Ultimaker/Cura/issues/14795
Sorry for the inconvenience.

closed 2023-04-05 by MariMakes