Ultimaker/Cura #15144: slicing failed

dkwolf1313, 2023-04-05:
### Application Version

5.3.0

### Platform

Windows 11 AMD RX5700XT

### Printer

Longer LK5 Pro

### Reproduction steps

Loaded STL file and setup parameters.
Clicked on Slice
Waited 10 minutes
Got error message that an unexpected error occurred and to consider making a bug report

### Actual results

No G Code created

### Expected results

G Code file Created

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

https://we.tl/t-j11Sxx2XQZ

rburema, 2023-04-13:
We've noticed an uptick in crashes. We think it might be Arachne related. Usually you can still get things to slice when you rotate the models by a little bit.

Since the download expired... Perhaps more importantly for diagnosing the problem, can you send us the relevant model(s)? (Saved project, not exported, you can zip the files to get github to accept them.) This way, it will also not expire.

rburema, 2023-04-13:
Thanks, I've added this issue to our internal ticket [CURA-9558](https://ultimaker.atlassian.net/browse/CURA-9558), which seems to be related.