Ultimaker/Cura #15149: Can't change G-code Flavor in 5.3.0

doubletrouble023, 2023-04-05:
### Application Version

5.3.0

### Platform

windows 10

### Printer

voron 2.4

### Reproduction steps

Previously I've been jumping back and forth between Marlin and RepRap flavor. ( I'm running Klipper )
It is set to RepRap now and when I try to change it, it just reverts back again.

### Actual results

Can't slice a file with another flavor anymore, tried saving, restarting etc. 

### Expected results

Cura should change the flavor, slice gcode files in that flavor and stay that way until told otherwise.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[](https://community.ultimaker.com/topic/43213-cant-change-g-code-flavor-since-530/)

MariMakes, 2023-04-05:
Hey @doubletrouble023,

Welcome to the UltiMaker Cura Github 🚀 
Thanks for your report 👍

This seems to be a duplicate of https://github.com/Ultimaker/Cura/issues/14795
This has been resolved and the fix will be available in an upcoming patch release with the correct translations.

I'll be closing this issue as a duplicate but you can follow the progress here: https://github.com/Ultimaker/Cura/issues/14795
Sorry for the inconvenience.

closed 2023-04-05 by MariMakes

doubletrouble023, 2023-04-05:
Oh! Sorry about that, I looked through a lot of the issues to see if I
could see it already posted somewhere. But apparently I didn't look far
back enough. Thanks!

MariMakes, 2023-04-06:
Hey @doubletrouble023,

Don't worry about it not seeing the duplicate.
It's easy to get lost in the list.  I'm glad you tried. 😊