Ultimaker/Cura #15150: cura crashes at 0.1mm layer height

adoeller, 2023-04-05:
### Application Version

5.3

### Platform

windows 10

### Printer

Tevo Tornado

### Reproduction steps

slicing with exact 0.1 layer thickness.
any other layer hight works

### Actual results

crash

### Expected results

guess

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log.zip](https://github.com/Ultimaker/Cura/files/11161675/cura.log.zip)

rburema, 2023-04-13:
Does this happen for _all_ models at 0.1mm, or just specific ones?

We've noticed an uptick in crashes. We think it might be Arachne related. Usually you can still get things to slice when you rotate the models by a little bit.

Perhaps more importantly for diagnosing the problem, can you send us the relevant model(s)? (Saved project, not exported, you can zip the files to get github to accept them.)

adoeller, 2023-04-13:
it crashes with this particular model.

rburema, 2023-04-13:
Can you please upload the model? We might have more of a chance to analyse it that way.

(Save as project, not export. Zip it in order to get github to accept it.)

adoeller, 2023-04-20:
does not happen any more with the latest version

rburema, 2023-04-21:
Glad to hear it! I'm closing this issue then.

closed 2023-04-21 by rburema