Ultimaker/Cura #15151: Cura will not save setting for gcode

brforsythe, 2023-04-05:
### Application Version

5.3.0

### Platform

Windows 11

### Printer

Ender 3 Pro

### Reproduction steps

1. made changes to settings for bed temperature and nozzle temperatures.
2. saved project
3. sliced to gcode
4. the gcode doesn't have the saved setting.
5. unistalled software
6. reinstalled software
7. same results

### Actual results

The gcode doesn't reflect the changes made in cura.

### Expected results

gcode should reflect the settings after slicing file

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[CE3PRO_ItsLitho_Image-T2nd.zip](https://github.com/Ultimaker/Cura/files/11161750/CE3PRO_ItsLitho_Image-T2nd.zip)

rburema, 2023-04-13:
After downloading your file, when I change the Bed Temperature and Initial Bed Temperature, I get M190 and M140 command with the proper temperatures (I selected) in the proper places for Cura 5.3.0

Could you maybe attach the gcode as well?

MariMakes, 2023-05-04:
Hey @brforsythe,

Welcome to the UltiMaker Cura Github 🚀 
I'm sorry to see you had to reinstall the version of Cura without success. 

I observe the following;
These are the temperature settings that are included in your project file
[image]

These are the temperatures that are sent to the slicer:

- material_print_temperature: 205.0
- material_print_temperature_layer_0: 205.0, 
- material_bed_temperature: 70, 
- material_bed_temperature_layer_0: 70, 

These are the temperatures that are in your Gcode:
```
M190 S70 ; Start heating the bed, wait until target temperature reached
M109 S205.0 ; Finish heating the nozzle
M104 S205
```

To me they all seem to be the same. 

@brforsythe, can you share how you observed the temperatures were different than you expected because I cannot reproduce your actual and expected behavior.

closed 2023-05-30 by github-actions[bot]