Ultimaker/Cura #15152: shutter on walls - no continuous speed

Hundsbuah, 2023-04-05:
### Application Version

5.3.0

### Platform

Windows 11

### Printer

PrincoreONE CoreXY with Klipper

### Reproduction steps

slice the attached piston stl with my attached cura 5.3.0 profile and print it or check it with https://www.gcodeanalyser.com/
did someone else encountered also such a problem?

### Actual results

shuttering / low velocity at the walls which lead to a bad surface finish

### Expected results

no shuttering, smooth surface finish like in the cura preview window

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[cura_5.3.0_profile.zip](https://github.com/Ultimaker/Cura/files/11162050/cura_5.3.0_profile.zip)
[piston_stl.zip](https://github.com/Ultimaker/Cura/files/11162054/piston_stl.zip)
[image]
[cura.log.zip](https://github.com/Ultimaker/Cura/files/11162104/cura.log.zip)
[CFFFP_píst, piston.gcode.zip](https://github.com/Ultimaker/Cura/files/11162105/CFFFP_pist.piston.gcode.zip)
[CFFFP_píst, piston_cura_project_file.3mf.zip](https://github.com/Ultimaker/Cura/files/11162107/CFFFP_pist.piston_cura_project_file.3mf.zip)

colatuttorf, 2023-04-06:
I am having this same issue on curved surfaces in 5.3.0 that I was not having in 4.9.1

[image]

rburema, 2023-04-13:
There seems to be micro-segments in the output messing this up. For the moment, it seems the best solution to avoid this is to downgrade to 5.2.2 -- I think the problem is known, but I'll bring it up with my colleagues as well to be sure.

jellespijker, 2023-05-03:
unfortunately not, there are multiple things in play here. This is just one of the smaller things that could help with better results.

Still in the process of investigating this.

MariMakes, 2023-05-04:
Hey All, 

This seems to be a duplicate of https://github.com/Ultimaker/Cura/issues/14811
@jellespijker is working on a solution for this. 

I'll be closing this issue as a duplicate but you can follow the progress here: https://github.com/Ultimaker/Cura/issues/14811