Ultimaker/Cura #15154: Support for custom G-code macros on layer change

ansonl, 2023-04-06:
### Is your feature request related to a problem?

Are there plans for Cura to support custom G-code macros on layer change? So that users can specify custom gcode containing brackets with calculations to insert on layer change for multiple purposes such as changing input shaping parameters on the fly.

Marlin now has both [Input Shaping](https://marlinfw.org/docs/gcode/M593.html) and [Fixed-Time Motion](https://marlinfw.org/docs/gcode/M493.html) with support for gcode that changes the frequency. 

### Describe the solution you'd like

Add an option to specify custom gcode with bracket expressions to run on layer change. Custom gcode may look like (taken from Marlin docs):

```
M593 F{(layer_num < 2 ? 0 : 15 + 45.0 * (layer_num - 2) / 297)} ; Hz Input Shaping Test
M493 A{(layer_num < 2 ? 0 : 15 + 45.0 * (layer_num - 2) / 297)} ; (Hz) X Input Shaping Test
M493 B{(layer_num < 2 ? 0 : 15 + 45.0 * (layer_num - 2) / 297)} ; (Hz) Y Input Shaping Test
```

### Describe alternatives you've considered

Prusaslicer supports custom gcode macros on layer change but they lack other functions of Cura that I would like to use at the same time. 

### Affected users and/or printers

All users running Marlin or Klipper.

GregValiant, 2023-04-06:
I was thinking that this would be easy to do with a post-processor.  Following your link to MarlinFW the M593 command states:
_"In Cura, you’ll first need to add the .py file in [this repository](https://www.github.com/dsdanielko/cura-ringing-tower-script/) to your Cura “scripts” folder. In Cura, enable Spiralize Outer Contour and set Minimum Layer Time to 0. Then go to Extensions > Post Processing > Modify G-Code. Click Add a script in the window that opens and use the dropdown menu to find Input Shaping. The default settings will let you test a range of 15Hz to 60Hz."_
Following the "this repository" link there is indeed a post-processor written for this.  There isn't a pull request for it yet, but I would expect to see one.

ansonl, 2023-04-11:
@GregValiant Thanks for the find, they must have just added the Cura info last week. I'll give it a try next chance I get.

GregValiant, 2023-04-11:
I played with the script (just to see what it does) and the numbers go well beyond 60.  On the one model I used it on it went all the way to 300.  My expectation was that it would go between 15 and 60 with an increment on each layer.

ansonl, 2023-04-16:
I tried the Cura plugin and it works for M593 input shaping. The layer height needed to be set to 0.2mm so that fewer frequencies are stepped through 15-60 to prevent it from going over 60. This seems to be a bug because the plugin expects higher layer heights and has an "end" frequency yet doesn't actually scale the frequency for the height of the object.