Ultimaker/Cura #15158: Removing SeeMeCNC

PartDaddy, 2023-04-06:
### Is your feature request related to a problem?

Hi, I'm the founder of SeeMeCNC. I was told by Makerbot support to post here, which seems odd. I can be reached at steve@ seemecnc .com

Our SeeMeCNC machines and profiles within Cura setup are outdated. This leaves new and old customers confused and creates additional customer service issues for SeeMeCNC. For now, we only support Cura 4.6.2. Later Cura releases stopped supporting single head, dual extruder, machines. 

### Describe the solution you'd like

Removing all SeeMeCNC printers from the list of "Add Printers" in Cura.  

We continually update our Cura setup guide to give our customers the best experience with Cura 4.6.2 and SeeMeCNC printers. 

### Describe alternatives you've considered

None. Removing default SeeMeCNC profiles which cause users a lot of problems and failure trying to use Cura software. 

### Affected users and/or printers

All users, all SeeMeCNC printers. Basically, the user gets a bad experience with our printer and Cura software.

rburema, 2023-04-13:
Hi! This is indeed the right place to request such things.

I've added it to the list of things to discuss, since while your request seems reasonable, _removing_ definitions is rare (but not unheard of), and I'd like to check with the rest of the devs whether we need to do anything special for that. (For instance, I can imagine a scenario where a user that has a SeeMeCNC machine active when upgrading to a newer version of Cura might not go well.)

Also we should confirm with our support people that they indeed have received such a request -- can't be careful enough these days.

PartDaddy, 2023-04-13:
Hi

Ok, thank you.

You can also verify me by emailing us through our seemecnc .com website. Or
***@***.***

Once in a while we update the printer settings and slice settings. Our
company setup guide isn't changed very often, but our github files are
modified more frequently.

Steve

rburema, 2023-04-21:
Sorry for the wait. As you've maybe noticed, I've contacted your company, and got a proper reply.

As it turns out, after discussion with some of my colleagues, we've never had to outright _remove_ devices from all of those available before. There is at the moment no script to handle users that install a new version of Cura over an old one, with one of the removed machines in their list of added printers -- especially if that machine is active. (It originally sounded simple to me too, but it isn't as it turns out.)

Instead we'd like to offer either of the following alternatives:
- We don't think later versions of Cura 'stopped supporting single head, dual extruder, machines' in general at least, so we'd like to know what goes wrong there, and perhaps fix it in the next version.
- If this is not acceptable, we'd like to instead build a small feature in the next version to let the user know that the use of a printer is deprecated, which would show the latest supported version (4.6.2 in your case).

MariMakes, 2023-06-07:
Hey @PartDaddy,

We see that about 20 users still use this printer in versions later than 4.6.
We don't want their cura to crash if they update to a new version of Cura. 

There is one low-risk option we can implement for you. 
We can add {DEPRICATED] in front of your printer names, so users know to seek support. 

I'll try to find some time tomorrow to update these two printer names:
https://github.com/Ultimaker/Cura/blob/main/resources/definitions/seemecnc_v32.def.json
https://github.com/Ultimaker/Cura/blob/main/resources/definitions/seemecnc_artemis.def.json

closed 2023-06-08 by github-actions[bot]

PartDaddy, 2023-06-08:
Hi guys

It's ok, let's not spend anymore time on it. You guys have more important
things to do.

I'm thankful Cura software has been free all of these years. It does a
remarkable job.

Thank you
Steve Wygant, CEO
SeeMeCNC
Blackpoint Engineering,  LLC

MariMakes, 2023-06-08:
Hey @PartDaddy,

Thanks for the reply. 
I'll close the issue than without changing anything. 👍

tmbdev, 2023-07-03:
The default Artemis profile that ships with Cura works fine, and I strongly recommend continue shipping it.

The official SeeMeCNC profiles are a mess, with separate profiles for different materials, single/dual setups, etc., ignoring the configuration logic and UI of Cura. My guess is that when @PartDaddy says that "Cura releases stopped supporting single head, dual extruder, machines", he is really talking about a problem with the way they ship profiles, not actually a problem with Cura dual extruder support or slicing.

SeeMeCNC needs to get their act together and submit and maintain an official, correctly written profile to the Cura repo. It is incomprehensible to me that this hasn't happened for a $2000 3D printer.