Ultimaker/Cura #15161: Constant Crashing at Random Times

alexp9982, 2023-04-06:
### Application Version

5.3 beta - 5.3

### Platform

macOS 13.2 - macOS 13.3

### Printer

Ultimaker S7, Ultimaker S3

### Reproduction steps

1. Open Cura
2. Wait until application crashes (times seem to be insanely random)
3. Application randomly crashes (and quite often), I do not see a pattern, sometimes a model will be loaded in, sometimes not; sometimes I will be actively working in Cura, sometimes not

### Actual results

Cura crashes and a message pops up saying "Cura has exited unexpectedly" requesting me to Ignore, Report to Apple, or Reopen

### Expected results

I should be able to have Cura run both as the current window and in the background without crashing.

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

The issue started happening in 5.3 tree support beta, and I thought that when 5.3 released, the issue would be fixed, but it is not. Let me know if I need to include the cura.log as well (I had to restart Cura to open folder, so the log ended up being erased

Crash log as reported by macOS has been attached below (the log that opens when you hit Report):
[curacrashlog.pdf](https://github.com/Ultimaker/Cura/files/11173885/curacrashlog.pdf)

alexp9982, 2023-04-07:
Crashed again, so here's the Cura.log file:
[cura.log](https://github.com/Ultimaker/Cura/files/11175760/cura.log)

alexp9982, 2023-04-07:
Leaving another Cura.log just in case:
[cura.log](https://github.com/Ultimaker/Cura/files/11181687/cura.log)

nallath, 2023-04-11:
I don't really see anything that catches my attention right away that could explain this. The one thing that I do see (but we haven't had any reports regarding that), is that you seem to put your machine in standby when Cura is running.

We are testing with a newer version of QT, which might also solve this issue, but we haven't created a version of that nightly build for macOS.

alexp9982, 2023-04-11:
It does indeed seem to be more common when the my Mac is waking up, come to notice. However, this problem doesn't appear to happen on my Windows laptop or PC. Seems sort of strange. Thanks for your reply

nallath, 2023-04-11:
If I've learned one thing from developing applications on multiple platforms is that they are different in all the weird ways that you don't want them to be different.

We've had issues with network requests causing issues on mac in combination with sleep, which is why I was looking for it. But those issues were quite some time ago, so I'm not sure if that's what is happening here.

alexp9982, 2023-04-11:
The logs always seem to end with a 403 error from the internet. It has indeed happened a few times without my MacBook going to sleep (including while actively using Cura), but the majority of the time, its within a minute from waking the screen back up.

This only started being an issue in the Tree Support alpha/beta from Christmas time. I'm not sure how Cura's Application Support folder is fully structured, but if they use the same "5.3" folder, with the same settings, maybe the Xmas version's settings are the same as the 5.3 Stable, which ends up causing the same problem as the Xmas version caused? I'm just tossing ideas out, since I don't know the inner workings of Cura. 

Although less likely, this could be caused by some sort of conflict in Rosetta, since Cura does not run natively on Apple Silicon Macs (not sure if this is intended or not, since I thought official M1 support was one of the big features of 5.0).

nallath, 2023-04-12:
Support is a big word. Before the big 5.0 release, even with rosetta, Cura didn't run properly. With the 5.0 release it should work, but it's not natively running on the M1. 

I think that it has something to do with Qt (the framework that we use for Cura), which also handles the network requests. I don't think that it's caused by the config files.

rburema, 2023-04-12:
If it _is_ Qt related, it _might_ be solved by the upgrade to Qt 6.4 we'd like to make for Cura 5.4 -- You can find a alpha for that here: https://github.com/Ultimaker/Cura/discussions/15098  (beware of bugs obviously). As far as I remember, this was also one of the prerequisites for native M1 support.

alexp9982, 2023-04-12:
Seems that there isn't a macOS version of this alpha build, but I would definitely try it out if I could.

rburema, 2023-04-13:
I've rebuild it for mac, can you use these links?:
- [dmg](https://github.com/Ultimaker/Cura/suites/11920862212/artifacts/644990118)
- [pkg](https://github.com/Ultimaker/Cura/suites/11920862212/artifacts/644990119)

alexp9982, 2023-04-13:
Unfortunate news: I received the first crash of the day. Once again, this was within 2 minutes of wake, according to the second log here. Weirdly enough, I had both the Stable and this test version up at the same time, and only this alpha version crashed. 

Here's the log file: 
[cura.log](https://github.com/Ultimaker/Cura/files/11225667/cura.log)

macOS crash log:
[curacrashlog.pdf](https://github.com/Ultimaker/Cura/files/11225683/curacrashlog.pdf)

rburema, 2023-04-14:
@alexp9982 Thanks for the investigation! Now that we have a clear cause, I can add it to the list of things to be discussed.

(P.S. Most versions of windows do have an open file limit one way or the other, it's just a bit/lot higher. Depending on what you're doing, anywhere from 512 to 2^16.)

MariMakes, 2023-05-04:
Hey @alexp9982,

Sorry to see you are having so many crashes 😞 

Thanks for the extra investigation ❤️ 

I'll bring it up to the team to see what they can do to improve it. 
Fingers crossed 🤞

MariMakes, 2023-05-08:
Hey @alexp9982,

I've discussed your issue with the team your extra investigation definitely helped with the discussion. Thank you 😊 
We expect that it might be related to icons incorrectly being incorrectly considered an "opened document". 

We've added a ticket to the backlog with the intent to improve this. 
For internal reference CURA-10581

Thanks for the report! 👍

cbagwell, 2023-05-30:
I came across this issue trying to find why Cura crashes on MacBook.  After reading this, it did make me realize it seems to only do it after resuming from a prolonged sleep.  I see same cloud login timeouts in my Cura log as well.

In case it helps, here is example of what is growing from "lsof -p <cura_pid> > curafilesX.txt".  First is from output after running for an hour but no sleep and shows 14 opened PIPE file descriptors.  The next is after resuming from prolonged sleep and shows 188 opened PIPE file descriptors and I'm approaching my limit of 256 FD (the last pipe is using the 212'th FD in this case).  

My Cura instances did crash about 10 minutes after I ran the second lsof command.  From Cura log, it appears to attempt a new web access every 1 minute and I assume leaking a PIPE each time?  

```
2023-05-30 07:20:50,523 - DEBUG - [MainThread] cura.OAuth2.AuthorizationService.
refreshAccessToken [174]: Was already busy refreshing token. Do not start a new 
request.
2023-05-30 07:20:50,528 - DEBUG - [MainThread] cura.UltimakerCloud.UltimakerClou
dScope.requestHook [30]: User is not logged in for Cloud API request to https://
api.ultimaker.com/connect/v1/clusters?status=active
```

```
$ grep "PIPE " curafiles1.txt | wc -l
      14
$ grep "PIPE " curafiles2.txt | wc -l
     188
$ grep "PIPE " curafiles2.txt | tail -1
UltiMaker 38252 chrisbagwell  212      PIPE 0x517a8f405e513b0d     16384                     ->0x34777435553bd82a
```

nallath, 2023-08-04:
Bit of an update on this: I've also done some digging and it seems to be a osx only thing. By default linux seems to have unlimited open files. Since I'm on linux, i tried the (very nicely written!) steps as described by @cbagwell. The open pipes don't seem to be an issue on linux as the number is always 0. 

Even keeping Cura open for a long time on linux, the number of open files never goes past 80.

cbagwell, 2024-04-19:
Update to my past info.  I have upgrade to MacOS Sonoma 14.4.1 and Cura 5.7 so I thought I see if that helped anything.

I am no currently able to reproduce the crash.  Looking at the # of open PIPE's in Curra, it would seem I'm over the previous 256 limit yet its not crashing:

```
$ lsof -p 46873 | grep PIPE | wc -l
     392
```

Looking at launchctl's values, it still reports 256 maxfiles can be opened by default (soft limit is first):

```
$ launchctl limit maxfiles
	maxfiles    256            unlimited 
```

This seem to be new.  ulimit now reports I can open 2560 max files.  I can't find much info on internet to indicate Apple increases the default but something appears to have adjusted my default max. 

```
$ ulimit -a | grep open
open files                          (-n) 2560
```

This might indicate the issue can be closed if it can be confirmed newer MacOS versions have increased the default limits.