Ultimaker/Cura #15162: Error during slicing

chaskins81, 2023-04-06:
### Application Version

5.3.0

### Platform

Windows 11

### Printer

Anycubic Mega X

### Reproduction steps

Received and error while slicing notification.
[image]

### Actual results

file wont slice

### Expected results

file slices...

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

not sure what else to add

November1234, 2023-04-07:
same problem dude....

MariMakes, 2023-05-04:
Hey @chaskins81,

Welcome to the UltiMaker Cura Github 🚀 

Have you tried reverting to the default settings?
Have you tried moving the model away from the center of the buildplate and rotating by 90 degrees?
Can you check if your mesh is faulty with the Meshtools Plugin? You can get it here: https://marketplace.ultimaker.com/app/cura/plugins/fieldofview/MeshTools

If that's not it, I'll need a little bit more information from you.
Do you have a project file for us? It contains the printer and settings we need for troubleshooting. 
To save a project file go to File -> Save project. Please make sure to .zip your project file. For big files, you may need to use WeTransfer or similar file-sharing sites.

You'll need to open Github in your browser to attach any files to your ticket
You can click the **view it on GitHub** link at the bottom of your email to open GitHub in the browser.

closed 2023-05-30 by github-actions[bot]