Ultimaker/Cura #15166: Im using this slicer for a Peopoly Moai 130, im running into "Slicing Failed" when im wanting to slice certain files, plz help

November1234, 2023-04-07:
### Application Version

5.3.0

### Platform

windows 11

### Printer

Peopoly Moai 130

### Reproduction steps

I will try to slice a stl file to get ready to print

### Actual results

when i try and slice it, i get a slicing failed, there was a unexpected error

### Expected results

the same result ive been getting

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[image]
[image]

November1234, 2023-04-07:
i also tried downloading the "newest" verision and nothing seems to work now, slicing wise

November1234, 2023-04-07:
i tried changing angle and position on buildplate still nothing, i used recommended plug in such as mesh settings still nothing, some files will slice while others wont

November1234, 2023-04-12:
why is there no support here?

rburema, 2023-04-13:
Hi, we're dealing with a high number of incoming requests right now.

We're currently investigation the slicing errors as well.

As a workaround, rotating the model slightly often seems to work.

rburema, 2023-04-13:
Additionally, if it's not for certain models but for everything, even simple shapes, it would be good to have the logs as well, as something else may be wrong.

MariMakes, 2023-05-04:
Hey @November1234,

Welcome to the UltiMaker Cura Github 🚀 
Sorry, it took us a while to get back to you 😞 

We are missing some information for troubleshooting. I can slice a simple cube in Cura 5.3.1 and your printer. 
[image]

Have you tried reverting to the default settings?
Have you tried moving the model away from the center of the buildplate and rotating by 90 degrees?
Can you check if your mesh is faulty with the Meshtools Plugin? You can get it here: https://marketplace.ultimaker.com/app/cura/plugins/fieldofview/MeshTools

If that's not it, I'll need a little bit more information from you.
Do you have a project file for us? It contains the printer and settings we need for troubleshooting. 
To save a project file go to File -> Save project. Please make sure to .zip your project file. For big files, you may need to use WeTransfer or similar file-sharing sites.

You'll need to open Github in your browser to attach any files to your ticket
You can click the **view it on GitHub** link at the bottom of your email to open GitHub in the browser.

closed 2023-05-30 by github-actions[bot]