Ultimaker/Cura #15167: Insert stl file into open project

fastStevie80, 2023-04-07:
### Application Version

5.3.0

### Platform

Windows 10

### Printer

Anycubic Kobra Neo

### Reproduction steps

1. I opened a stl file with Ultimaker Cura 5.3.0
2. I imported another stl file
3. 

### Actual results

the two different stl objects were kinda "morphed" into one
[image]

### Expected results

The two stl objects should be placed next to another
[image]

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[CFFFP_OttoDIY_feet_V11.zip](https://github.com/Ultimaker/Cura/files/11177862/CFFFP_OttoDIY_feet_V11.zip)
[cura.log](https://github.com/Ultimaker/Cura/files/11177866/cura.log)

fieldOfView, 2023-04-07:
In the Cura preferences, do you have the "Ensure models are kept apart" setting checked or unchecked?

fastStevie80, 2023-04-07:
Hello,

thank you very much for this quick response on a holiday...
I didn´t have set this. I never had changed this setting. It worked fine
till the update from Cura 5.2.2 to Cura 5.3.
Maybe it got changed automatically...

Again: Thank you very much and have a good time.

Many greetings!

closed 2023-04-07 by fastStevie80