Ultimaker/Cura #15169: Unexpected error when slicing

zumbra, 2023-04-07:
### Application Version

5.3.0

### Platform

Windows 10

### Printer

Ultimaker S5

### Reproduction steps

After some minutes of slicing the sw get the following error:
SLicing failed with an unexpected error. Please consider reporting a bug on our issue tracker.
I know the file is complicated but I don't understand what I can do to solve the issue as it's not giving me any clue.

### Actual results

After some minutes of slicing the sw get the following error:
SLicing failed with an unexpected error. Please consider reporting a bug on our issue tracker.
I know the file is complicated but I don't understand what I can do to solve the issue as it's not giving me any clue.

### Expected results

After some minutes of slicing the sw get the following error:
SLicing failed with an unexpected error. Please consider reporting a bug on our issue tracker.
I know the file is complicated but I don't understand what I can do to solve the issue as it's not giving me any clue.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[UMS5_Ranieri_zed2_Mix_KO.zip](https://github.com/Ultimaker/Cura/files/11178450/UMS5_Ranieri_zed2_Mix_KO.zip)

stewwy, 2023-04-07:
FYI This happened to me, tried slicing mulitiple times and failed ~30% of the way, moved the model on the bed 4mm and it re-sliced OK

zumbra, 2023-04-11:
Wow, seems incredible but it worked!!! Thank you, it would have never came in mind such a solution.
Thank you very much. It was precious.

jellespijker, 2024-01-14:
Thank you for your report. It seems that your uploaded project file slices correct on our latest unreleased 5.7.0-alpha.0+3abb5 which indicates that the issue you're experiencing seems to be fixed for the upcoming 5.7.0 release.
I'm closing this issue as fixed for the 5.7.0 release
[image]

closed 2024-01-14 by jellespijker